Ralim/IronOS #83: License

numinit, 2017-09-08:
Hate to be that guy, but is a license for this project listed anywhere?

Ralim, 2017-09-08:
No problem being that guy,
I do not know a fully correct answer to that question as the original source is (as far as I know) not give a licence. As this firmware 1.x fork did ultimately come from that code I do not feel comfortable pinning a specific licence on the code. However, as far as I am concerned I treat the code as being free to use however you see fit in your application.

There is a coming re-build of this code (version 2.x), that is a complete re-build that shares no real code or logic from the original code. For this, I will probably need to select an appropriate licence.

Not sure that overly helps, is there a specific use case you are concerned about?

tigert, 2017-09-08:
I cannot speak for @numinit but having a clearly defined license would make things more clear for contributions – you know what you can do with the codebase and how your contributed code can be used.

Ralim, 2017-09-08:
That makes sense, I just couldn't really feel how to settle on one considering the lack of direction from the original manufacturer. Would you have a preferred licence you prefer to work with that you would suggest in this situation?

numinit, 2017-09-08:
@tigert Pretty much what I was after :-)

@Ralim How much of the code is copied directly from the original firmware?

Ralim, 2017-09-08:
Yeah that's fair enough,
At this point, the current code would probably be MIT style, though I'm hesitant to put a licence on the current code. In 1.x very, very little of the code is left except the setup code now along with the idle screen graphics. However, I am adverse to picking one out specifically since it is a Chinese product and they do not use a licence style system see "Gongkai" [as bunnie put it](https://www.bunniestudios.com/blog/?p=3040).

In 2.x there is no original code left except the idle screen graphics, which might be going. Note that 2.x moves to the newer stm hal libraries, I know they are not as efficient but they are easier for some users. These are licenced as BSD, however, I would be most likely to use GNU GPL v3 for the code.

Thoughts ?

numinit, 2017-09-08:
Personally, I'm a fan of the GPL, as far as keeping the source "free and open" goes. It's rare to see firmware with such a license, especially considering that it would otherwise be binary blobs or not provided entirely.

EricJanVanDenBogaard, 2017-09-08:
The harware of the TS100 is Open Source, that's the nice thing of it, and this software is also openly shared for everybody to use or modify, to make it better, closed/copyrighted software is a bad thing for the community, Open Source is licenced to everybody by the correct GPL rules, keyword Arduino :)

numinit, 2017-09-08:
@EricJanVanDenBogaard well, open source isn't _exactly_ the same as free and open source. The GPL is an infectious license, so anything that's derived has to be made available under the GPL.

For example, it would be a shame if a knockoff of the TS100 was made with closed-source firmware derived from this repository. The GPL ensures that, if this happens, you at least have some of the standing to request that the source be made available.

tigert, 2017-09-08:
so, I am fine with any open source license. GPL is kinda 

I can redraw the graphics if that helps, though you'd need to convert tgem to the correct format.

Ralim, 2017-09-10:
@numinit I agree about the knockoff iron idea, which is why for the 2.x branch I will be adding a GPL licence to the code.

@tigert If you do happen to feel like making images I might take you up on that soon :)

tigert, 2017-09-10:
Sure, I can do the bitmaps.

I assume you would be coding them into header files by hand? 

I guess I would need a list of images in use and in particular the pixel sizes? 

Battery images come to mind, also heating and cooling and steady? Settings too, where one probably should separate the image of the iron with two buttons from the labels /arrows.

Anything else?

I can file a separate issue about the graphics.

EricJanVanDenBogaard, 2017-09-10:
thought of GPL as a thing you get the code for free to work with, and if some one is able to make it better, everybody has profit from that. otherwise you get Raceflight drama's 
btw the hardware is also under GPL ?

uncle-fed, 2017-09-12:
Make sure the fonts that you used for text characters are 1) allowed to be used in your project 2) compatible with the license you decide to go with.

Ralim, 2017-09-12:
@uncle-fed Please see my response to your comment on issue #61 .
As far as I am aware and can confirm the font is ok to be used in this way.

Ralim, 2017-09-28:
The newer code is GPL V3, and so I'm going to close this as the 2.x fork should be becoming the main release soon (hopefully)

closed 2017-09-28 by Ralim