Ralim/IronOS #85: Thermal run away

dchosking, 2017-09-10:
I am having an issue wich seems like a thermal run away. I flashed the ralim firmware and the iron seemed to perform nicely. I then whent through the settings and changed a few

What I can remeber changing is the following.
1) set power mode to dc
2) set boost temp to 450

I then changed to a different tip and attempted to enter the calibration mode by pressing both buttons simultaneously.

The iron immediately started to heat up I only noticed it when it turned a bright glowing red it got so hot.

I assuemd maybe it was a broken tip temperature sensor so I replaced the tip with the previous one that was working fine. As soon as plugging in the iron now it starts to heat up and does not stop I dont even need to turn it on.

I attempted to reflash the firmware hoping to have the settings be reset but to no avail, the iron just gets hotter and hotter once plugged in.

Ralim, 2017-09-10:
Hi,
This is a serious problem and should not have occurred. It honestly sounds more like a hardware change that has caused this rather than software. If the tip is heating without even turning on the soldering mode that means that the output mosfet is latched on for one reason or another.

Can you confirm where you bought the iron from and also if the issue persists with the original firmware.
You can reset the settings by holding the rear button when powering on (the one you *dont* use for entering DFU, and keeping holding until it says reset ok).

That said the firmware does not change or update the heating timer until you enter the soldering mode so it should not heat like that at all, and the most likely cause is a hardware issue with the iron itself. Thus, if you could try the original firmware, and if that also causes the issue it is definitely a hardware issue with your iron. If it does not occur on the original firmware then I will make up a diagnostic firmware for your unit.

dchosking, 2017-09-10:
So I just did exactly that and flashed back the original firmware and it does the exact same thing. The only conclusion is that it has to be a hardware issue. It is really depressing as I used thr iron a total of once 😕

Ralim, 2017-09-10:
Yeah 😢 
The fact that it happened on the first try also makes it look like a hardware failure (I have abused mine quite a lot and not had it fail from *use*).

Did you buy yours as a "TS100" or as one of the rebrands? (If you don't mind saying?)

dchosking, 2017-09-10:
It is this one that I got from banggood
https://m.banggood.com/MINI-TS100-Digital-OLED-Programable-Interface-DC-5525-Soldering-Iron-Station-Built-in-STM32-Chip-p-984214.html?rmmds=myorders

awigen, 2017-09-10:
Hi dchosking,

It sounds like the output mosfet has shorted (CJQ4953). MOSFETs will often fail short. I'd start looking there.

Cheers,
Alex

EricJanVanDenBogaard, 2017-09-10:
btw... That font on that Banggood picture looks also different from the original software it has to start with ?
Are most components replaceable of the TS100 pcb ?

dchosking, 2017-09-10:
I am going to ask banggood for a refund I think, and if they dont then will maybe try fix it.
This is a quick video I made showing the issue.
https://youtu.be/pkZr3zR97O4

closed 2017-09-10 by dchosking

Ralim, 2017-09-11:
@awigen I agree it's most likely the output mosfet that has failed.

Let us know if you don't get any help from banggood.

chargerdude70, 2017-10-11:
I have the same issue with one straight out of the box. Heats to cherry red and shows sen-err. Tried extra tip I ordered with it and still no good

Ralim, 2017-10-11:
@chargerdude70 
Can you please verify if the same issue occurs with the original firmware? 
Did you test the original firmware first when you received the iron?

chargerdude70, 2017-10-11:
That was with original fw. After seeing that issue and reading this github thread....I did not bother to update the fw to your goodness.

chargerdude70, 2017-10-11:
@Ralim^^^^

Ralim, 2017-10-11:
@chargerdude70 It's weird that these issues are appearing suddenly. Can I ask where you picked up your iron from? Is yours also from the same Banggood listing?
If they are from the same source I'll let eDesign know there might be a production issue.

chargerdude70, 2017-10-11:
@Ralim 
ordered from banggood ..got it during their anniversary sale... they were out of stock so I had to wait for new stock....  just received it on 10-4 and started playing with it yesterday

chargerdude70, 2017-10-11:
@dchosking
I asked them to send a new one. I do not know how it's going to work out. I purchase it $20 less than what they're asking for now. I do not want a refund it's going to cost me $20 more. If that mosfet is bad I can replace for a $1.65.

chargerdude70, 2017-10-11:
@Ralim
Do you know if the mosfet can be diagnosed with the MCU daughter board removed? And if so would you know a procedure on how to do so?

Ralim, 2017-10-12:
@chargerdude70 
I would first remove power and the daughter board, and just test the Drain-Sink of the mosfet to see if its shorted. If it is then its most likely just the mosfet needing replacement.

If the mosfets not shorted, I would apply power but no MCU and see if it still overheats. If it does, then something else on the pcb has failed. I would then work backwards along the schematic to see where the signal is incorrect.

If it all looks good and doesnt randomly heat up, then something on the daughterboard is weird. Which might be a bit harder to diagnose. Though there really isnt much on there.

chargerdude70, 2017-10-12:
@Ralim 

BG's response to my issue. At work for next 12 hrs so wont know until much later if it's a fix or not. 

[image]

chargerdude70, 2017-10-13:
Updated FW to 2.17 per BG request and still no deal. 

https://www.youtube.com/watch?v=XJF1YH6FLMA

dchoskingsage, 2017-11-07:
So i got my replacement and two weeks later same thing happened, i am so frustrated i dont even want to try and ask for a third replacement (which i am yet to be refunded for he first one that broke, i just bought my own replacement) .
I suspect that the items from BG are possibly clones or part of a batch that has some production issue.

Interestingly enough both failures happened after swapping out to a different tip.

I think i would like to try fix this one i have now but have no idea where to start

chargerdude70, 2017-11-07:
I received a replacement for mine as well. It works as desired but I have yet to replace the tip for another so I cant verify as of yet if mine will do the same. BTW, mine were from BG as well.

dchoskingsage, 2017-11-16:
You guy seem to know what you talking about, I would like to try replace the mofset but cant get that exact same one. Do you think this one would suffice. 

https://www.mouser.co.za/ProductDetail/Vishay/SI4459ADY-T1-GE3/

Also sorry i am posting here i have no idea where else to get help from and i know this is not exactly the forum for this sort of support

Ralim, 2017-11-19:
Hi @dchosking 
That mosfet won't work as a direct replacement. You are looking for a dual P MOSFET.

https://www.mouser.co.za/ProductDetail/ROHM-Semiconductor/SH8J31GZETB/?qs=sGAEpiMZZMshyDBzk1%2fWi9lDZIJVG72oj5Al7AXZulLtfR8s%2fUIC2w%3d%3d

This one, for example, meets the same specifications and has the same pinout. 
Should be just a drop in replacement once you remove the original one.

chargerdude70, 2017-12-25:
I replaced the mosfet and all is well with received defective  unit.... Thanks for the guidance 👍

Ralim, 2017-12-26:
Awesome :)

dchoskingsage, 2018-01-12:
So i have been looking to do the same thing but after going to actually buy the replacement MOSFET from mouser i realized their shipping is insane it puts the cost to the point where its almost the cost for a brand new TS100..... I found a place that has cheaper delivery but i am not 100% certain i am looking at a compatible mosfet again as they dont have the one @Ralim suggested.

Could someone just verify for me if this one would work 
https://za.rs-online.com/web/p/mosfet-transistors/7879052/

it would be much appreciated

Ralim, 2018-01-15:
That mosfet looks like it should work fine for the iron :)
RS Components are great with shipping usually (I use them a fair bit)

JohnEdwa, 2018-01-17:
@dchoskingsage If you have a local [Element 14/Farnell](http://farnell.com/) distributor, you can order parts for pickup without any delivery charges.

dchosking, 2018-01-23:
@Ralim thanks for the help, replaced the MOSFET and its 100% again. Super happy with this fix !

tamasgal, 2018-03-08:
I had the very same problem. I contacted the producer support directly and they sent me a new one within two weeks. I plugged the new one in and it failed with the same error. I suspect that it might break related to the tip (I plugged in my old one).

I will replace the MOSFET and report back. Thanks so far!

Btw. here is a video demonstrating the problem: https://www.youtube.com/watch?v=z0skca3p5uQ
And two follow-up videos with the replacement unit: https://www.youtube.com/watch?v=xSNQSDJywtY and https://www.youtube.com/watch?v=WU34c7GJgyw

tamasgal, 2018-03-13:
So a quick update: I replaced the Q1 MOSFET on both of my TS100 and both are working fine again.

I made a video how to do the repair, if anyone is interested: https://www.youtube.com/watch?v=g-xXSJagz_A

And here is a list of replacement MOSFETs:

- HM4953A or CJQ4953 (used in TS100)
- SI4925DY (the one I used, with -6.9A)
- SI4925DDY (the same as above, but -5A)
- SH8J31 (-60V, -4.5A)
- IRF7328PbF (-30V, -8A)
- IRF9362PbF (-30V, -8A)
- AO4801A (-30V, -5A)

tamasgal, 2018-03-14:
I am quite confident that I found the reason why some TS100 are failing with shorted MOSFETs:
1. A faulty tip can short out the MOSFETs source and drain pins, which will induce a very high current through it.
2. The current seems to be mostly limited by the power supply, which is often a simple laptop charger with a current limit protection of about 2.1A, which is within the specifications of the original MOSFET.

However, if you use a power supply with which can deliver more than 5A,  you risk that the maximal current (5A) of the MOSFET is exceeded and you grill it.

This is also the reason why I have chosen a MOSFET with -6.9A.

dchoskingsage, 2018-03-14:
@tamasgal I am fairly certain you are correct, i use a lipo which can supply upwards of 100amps with my TS100 and when i got the replacement it immediately failed after swapping out to the same tip that was on the previous iron that failed. I have replaced my Mosfet with an 8amp capable one now, so hopefully it is a bit more reliable now.

tamasgal, 2018-03-14:
Yep! Btw. the support sent me a mail a few hours ago (they found my video ;-) and I am now in contact with them. I don't have the schematics but I am sure they will include a current limiting component in the MOSFET circuit in the next version. At least I will try to convince them ;)

marvedjensen, 2018-07-09:
i have 2 irons with the same error   (sens error)as it call in the old firmware in the new firmware the error showing as (bad tip) my irons don't heat but do u guys think it also the mosfet

tamasgal, 2018-07-09:
Hey Martin, I just answered on your question on YouTube. Do you have another tip for test purposes? I guess it's just a broken tip. You can measure its resistance which I guess should be higher than 8 Ohm (this was the highest resistance I measured on a working tip).
I don't think it's the MOSFET.

marvedjensen, 2018-07-09:
ty Thomas I try and get back here : D

marvedjensen, 2018-07-09:
my  brand new tip is 8.1  and the 2 old ones I 8.0 and 7.5  ohms

tamasgal, 2018-07-09:
Then I'd say they are OK. It's a bit suspicious though that both of your TS100 failed. I'd first try to measure the voltage at the tip and do some general troubleshooting before you replace any parts. However, you can unsolder the MOSFET and check if source and drain are shorted (as I have explained and shown in my youtube video).

marvedjensen, 2018-07-09:
at the tip the volt is 3.98v the volt coming in the jack is  is 24v

marvedjensen, 2018-07-09:
3.9 on both my irons

marvedjensen, 2018-07-09:
the out pin 1 on the mosfet deliver 24v

tamasgal, 2018-07-09:
That should be OK (if you don't measure the same voltage on the drain pin ;). You can check the schematics to see if you find anything weird (image search for "ts100 schematics").

marvedjensen, 2018-07-09:
https://www.youtube.com/watch?v=-Zbb4WM-FCg&t=199s a little video of my 2 irons 

the problem is I really don't know what i should mesuere and what values is ok

  any tips of where to start measuring out the iron

tamasgal, 2018-07-09:
This is very weird... I also the quickly counting temperature readout is confusing me, on my irons, the values are stable when I turn on the iron.
What are the specs of your power supply?

You could check the temperature sensor, which is a TMP35:
http://www.analog.com/media/en/technical-documentation/data-sheets/TMP35_36_37.pdf

marvedjensen, 2018-07-10:
24v 5a  the sensor dont give det correct temp i put it on My Heather plate and it stay at 30c

Repled, 2018-07-24:
Hi All,

I have had all kind of issues. Bought 2 irons at once. Both had the latest Firmware (FW) 2.18 but none of them worked out of the box. Tried to download new FW to the units and after reading on the forum (http://www.minidso.com)  I got one working with reload of the offical 2.18 and the other with the latest Ralim. This took hundreds of tries to get it to work !!! ???

Tried to calibrate the input voltage to the iron with the Ralim FW. Got stuck, could change temp but not exit the calibration state. Pulled the power. After reconnection i got BAD TIP, and within a short while I had a glowing red tip !!! Broken Mosfet, now changed to a single DPAK P-channel Mosfet 60V, 15A, and working again :o)

1. Has anyone any ideas on why it is sooooo hard to get the FW into the units ?

2. This thread started with someone going into calibration and a broken Mosfet. My tip is not broken but the Mosfet got broken. Don't really see the failure mechanism. If the SW is doing something wrong and let's assume it pulls the Mosfet all the time, this should normally not kill the Mosfet. Over voltage ???

)) P