Ralim/IronOS #87: Russian / Ukrainian Translation

uncle-fed, 2017-09-11:
I can supply Russian / Ukrainian translation when / if you get the font ready.
Alternatively, I can try and supply Cyrillic font, if I can learn a bit more about how it is encoded in the source code. Can you provide direct hint regarding this?

Ralim, 2017-09-12:
That would be awesome,
I have a tool thats a modification of the boot up creator.
If you can provide a bitmap image for each symbol (16 high, ~11 wide).
 They are drawn as 12 wide but each digit has slight space around it.
Alternately if you can provide a bitmap font image (the grid table of all the letters) that is approximately the correct size, I can digitize that in as well.

uncle-fed, 2017-09-12:
Sure, I can provide the bitmaps. I can even digitise it too, so you have less work to do, if you tell me how it is encoded. BTW, Russian alphabet has 33 characters, 11 of which can be expressed with the existing Latin font, but 22 has to be added. Is there a limit to the characters you can have? If this becomes problematic, there are ways to work around the limitation.

Ralim, 2017-09-12:
There is no limitation to the characters I can add, as I'm just adding the extension characters to the font table as they would be in order in utf8 usually.
The digitisation is easy to do with software, but if you want to, the format is documented in another issue on here somewhere. I'll get a link shortly.

uncle-fed, 2017-09-12:
I could not find the format in the issues, but I already figured it out myself. Will work on the Cyrillic bitmaps now and probably will submit pull request.

Ralim, 2017-09-12:
Thanks, It's not a terribly hard encoding scheme if you have worked with LCD's in the past.

uncle-fed, 2017-09-13:
I've created this as a little helper. Perhaps it is also useful to you or others.
http://ipbox.linkpc.net/tsfed/

Ralim, 2017-09-13:
That's nicer than the tool I hacked together, thank you :smile:

uncle-fed, 2017-09-25:
While working on the Cyrillic font I have a question regarding the current Latin font.
The Latin font has been made so that for most capitalised characters and also numeric symbols have no empty space above them and 2 pixels of space below them. Is there a special reason to have it this way? I am asking this because a few Cyrillic (and also German / Italian / French) characters have special accents above them and there is no place to put those now, if you want to keep all characters in line with each other, i.e. equal height. Wouldn't it be more logical to have 1px above and below free, rather than 0px an 2px as it is now?

Ralim, 2017-09-26:
The current font is encoded that ways purely as that is how the original code I based off it is.
I have no reservation against changing the current font to be shifted up one pixel if that makes other glyphs easier to create 😄 

Also on this, any chance at all it would be possible to do an 8-pixel high version of these eventually as well (for 2.x detailed screens to be able to be translated)?

uncle-fed, 2017-09-26:
Sure, I can do 8-pixel high version. Is it still 12px wide?

Ralim, 2017-09-26:
At the moment its 5x8, for the current font, but it's not even full coverage of the larger font so not sure if I'll end up re-creating it. I would *like* to stick with 5 wide and 8 high, but it will come down to what is doable in fonts and what is just not possible. 

So more than open to ideas on this one as I have very little knowledge about what would work for you.

uncle-fed, 2017-09-27:
I've looked at a few Cyrillic characters that are most wide, like this ones: 
https://en.wikipedia.org/wiki/Nje#/media/File:Cyrillic_letter_Nje_-_uppercase_and_lowercase.svg
https://en.wikipedia.org/wiki/Shcha#/media/File:Cyrillic_letter_Shcha_-_uppercase_and_lowercase.svg

We most definitely would require 6px width. For most glyphs we should aim at using 5px with the rightmost 1px reserved for spacing between the characters. The very wide characters would use all 6px, luckily though, only few pixels (one or two) in the 6th "spacing" column would be needed, so they should not become glued to the characters following them too much.

uncle-fed, 2017-09-27:
Regarding my pull request https://github.com/Ralim/ts100/pull/98 that adds 12x16 Cyrillic characters.
I hope the current code handles double-byte UTF-8 encoded characters because you'll have to deal with that in the end. Cyrillic UTF-8 range is all double-byte where the first byte is always 0xD0 or 0xD1 (I made each characters code available in the comments).

Ralim, 2017-09-27:
@uncle-fed 
6-pixel width should be fine, it's currently using 6 but was hoping for 5. No real issue though 😄 

Yes currently it does handle double bytes, not particularly neatly but it works :) (2.x branch).

I'm hoping to be able to get a tester 2.x release up for you shortly :)

Thank you for marking the double bytes next to them all :)

uncle-fed, 2017-09-27:
Another thought for the 6x8 font: we should probably implement single case (uppercase or lowercase) only characters. It makes little sense having both upper and lowercase characters for menus / options with such a low-res screen. Also, if the vertical resolution of TS100 display is 16px, then we should reserve 1px for vertical spacing, so in the end our font will be 5x7 with 1 extra pixel to the right and bottom allowed for a few characters that really require it.

Ralim, 2017-09-28:
@uncle-fed 
I'm completely open to single case or both at this point honestly it's up to you 😄 
I agree that 5x7 makes the most sense as the actual character sizing if you can make it work?

uncle-fed, 2017-09-28:
Here's a link for some inspiration (watch for the licensing terms):

https://fontstruct.com/gallery/tag/386/5x7

uncle-fed, 2017-09-30:
According to this page: https://creativecommons.org/share-your-work/licensing-considerations/compatible-licenses/ the casionostalgia font license is not compatible with GPL.  :(

Ralim, 2017-10-02:
Damn :( The font not being GPL is a pain :/
If you do manage to find any compatible ones I would be seriously interested in supporting more characters :) Just don't feel confident doing my own interpretation of the glyphs as I don't really know what I'm doing with them 😄 

I'm still working on the Russian font working, by the way, just having some characters not decoding correctly at the moment. Hopefully, I'll be back soon with more news :)

uncle-fed, 2017-10-03:
Regarding the characters not working, do you mean that there are problems with the C code or did I get them encoded wrong? If its the programming code, I am curious why only some characters are misbehaving? Shouldn't the code be generic to treat all characters equally (at least within a certain range). Could I perhaps help with this issue?

Ralim, 2017-11-02:
@runy1984 
Time.
I just haven't had more time to work on other languages support in 2.x just yet.

uncle-fed, 2017-11-03:
What about Cyrillic support in 1.x ? The fonts are now available and also the translated strings.

Ralim, 2017-11-04:
Hi,
Can you try the attached file? I'm missing some translations atm, but if you could check the font first. Then we can do the translations.

[TS100_ru.zip](https://github.com/Ralim/ts100/files/1442891/TS100_ru.zip)

runy1984, 2017-11-04:
OK! check !

Ralim, 2017-11-04:
I've actually pushed out 2.x with the incomplete translations since its going to be the easiest way to get all of them updated. The above file is the same as the 2.x release. 

Let me know if they show okay, and also translations for the missing ones would be awesome :)

runy1984, 2017-11-04:
ОК!

uncle-fed, 2017-12-06:
This is an update on the CasioNostalgia 5x7 font located here:
https://fontstruct.com/fontstructions/show/1193786/casionostalgia

I got in touch with the author of the font [zdimension] and discussed the licensing issue with him.
He replied that the following: "I contacted FontStruct about the license and it seems CC4 is not supported yet (on the website). I will put a note on the page of the font saying that it's under CC version 4, that should be okay I guess. If you need anything else feel free to ask."

According to the information here BY-SA 4.0 is fully compatible with GPLv3.
https://creativecommons.org/share-your-work/licensing-considerations/compatible-licenses/

This means the font can be used in this project and we should go ahead with it in my opinion.
Let me know if you need any further input from me, otherwise I will stop looking into this issue for now.