Ralim/IronOS #90: Allow building the project using a Makefile

dogtopus, 2017-09-13:
Not all people wants to install and learn how to use System Workbench just so they can build this project. By adding a Makefile to the project, the build can be done by just typing a single make command (with the appropriate toolchain installed), which simplifies the things a lot. It also allows automated build which is crucial to several things (e.g. continuous integration)

Ralim, 2017-09-13:
Hi,
I agree I need to make a general one that is in the repository. At the moment I'm using the generated makefile for building everything.

I'll try to get this added shortly :)

thanks4opensource, 2017-09-13:
Yes! I've been wanting to request the same thing. Thanks in advance for offering to do this, @Ralim.

I'd like to try hacking on some of the outstanding issues -- like #65 ;) -- but am wary of doing so without easy debugging access. I'd have a lot more confidence if I could build the sources into a .hex file that's byte-identical to the distributed one, though any changes in compiler version or optimization flags make that unlikely. But without a Makefile to even a show the order in which the .o files are linked it's impossible. (BTW, thanks for including the LinkerScript.ld)

Maybe even a copy of the ELF format output, pre-conversion to .hex, that matches the sources if the System Workbench workflow creates one. Again, a byte-by-byte match will be difficult, but the metadata that's in a .elf would help identify any build-related problems.

Ralim, 2017-12-30:
@thanks4opensource 
@dogtopus 
Closing this as there is now a Makefile in the repository (and a bash script to run all the builds at once)

closed 2017-12-30 by Ralim