Ralim/IronOS #91: Russian language

runy1984, 2017-09-16:
please do Russian language !!!  all Russians will thank you !!!  Please !!!

Maartenwut, 2017-09-16:
You can do it yourself actually, take a look at https://github.com/Ralim/ts100/blob/master/workspace/ts100/src/Strings.c.

runy1984, 2017-09-16:
I don't understand! could you help me to do it please!  please!  please!

Maartenwut, 2017-09-16:
Could you translate there strings (text) to Russian? It's just the things in apostrophes.

#ifdef LANG_RU
const char* SettingsLongNames[14] =
		{
		/*These are all the help text for all the settings.*/
		/*All must start with 6 spaces so they come on screen nicely.*/
		"      Power source. Sets cutoff voltage. <DC 10V> <S 3.3V per cell>",
				"      Sleep Temperature <C>", "      Sleep Timeout <Minutes>",
				"      Shutdown Timeout <Minutes>",
				"      Motion Sensitivity <0.Off 1.least sensitive 9.most sensitive>",
				"      Temperature Unit", "      Temperature Rounding Amount",
				"      Temperature Display Update Rate",
				"      Display Orientation <A. Automatic L. Left Handed R. Right Handed>",
				"      Enable front key enters boost mode 450C mode when soldering",
				"      Temperature when in \"boost\" mode",
				"      Changes the arrows to a power display when soldering",
				"      Automatically starts the iron into soldering on power up. T=Soldering, S= Sleep mode,F=Off",
				"      Blink the temperature on the cooling screen while the tip is still hot." };

const char* TempCalStatus[3] = { "Cal Temp", "Cal OK  ", "Cal Fail" }; //All fixed 8 chars
const char* UVLOWarningString = "Low Volt"; //Fixed width 8 chars
const char* CoolingPromptString = "Off  "; //Fixed width 5 chars
const char SettingTrueChar = 'T';
const char SettingFalseChar = 'F';
const char SettingSleepChar = 'S';
const char SettingFastChar = 'F';
const char SettingMediumChar = 'M';
const char SettingSlowChar = 'S';
const char SettingRightChar = 'R';
const char SettingLeftChar = 'L';
const char SettingAutoChar = 'A';
const char SettingTempCChar = 'C';
const char SettingTempFChar = 'F';

#endif

runy1984, 2017-09-16:
ок!!!

runy1984, 2017-09-17:
#ifdef LANG_RU
const char* SettingsLongNames[14] =
		{
		/*These are all the help text for all the settings.*/
		/*All must start with 6 spaces so they come on screen nicely.*/
		"     Источник питания. Установка напряжения отключения. <DC 10V> <S 3.3 V на батарею>",
				"      Температура Сна  <С>", "    Переход в режим Сна <Минуты>",
				"       Переходит в режим ожидания <Минуты>",
				"      Акселерометр <0. Выкл. 1. мин. чувствительный 9. макс. чувствительный>",
				"      В чем измерять температуру", "      Шаг измерения температуры",
				"     Скорость обновления дисплея температуры",
				"      Ориентация Дисплея <A. Автоматический L. Левая Рука R. Правая Рука>",
				"      Активация  кнопки A для Турбо режима до  450С при пайке ",
				"      Установка температуры для Турбо режима",
				"      Изменяет стрелки на дисплей питания при пайке",
				"      Автоматический запуск паяльника при включении питания. T=Нагрев, S= 
Режим Сна,F=Выкл.",
				"      Мигает температура на экране охлаждения, пока жало остается горячим." };

const char* TempCalStatus[3] = { "Cal Temp", "Cal OK  ", "Cal Fail" }; //All fixed 8 chars
const char* UVLOWarningString = "Low Volt"; //Fixed width 8 chars
const char* CoolingPromptString = "Выкл.  "; //Fixed width 5 chars
const char SettingTrueChar = 'T';
const char SettingFalseChar = 'F';
const char SettingSleepChar = 'S';
const char SettingFastChar = 'F';
const char SettingMediumChar = 'M';
const char SettingSlowChar = 'S';
const char SettingRightChar = 'R';
const char SettingLeftChar = 'L';
const char SettingAutoChar = 'A';
const char SettingTempCChar = 'C';
const char SettingTempFChar = 'F';

#endif

Ralim, 2017-09-17:
Thank you,
This is partially a duplicate of #87.
This will also be pending waiting on a font map being made that contains the required characters.

runy1984, 2017-09-17:
what else to do?

runy1984, 2017-09-17:
#ifdef LANG_RU
const char* SettingsLongNames[14] =
{
/These are all the help text for all the settings./
/All must start with 6 spaces so they come on screen nicely./
" Источник питания. Установка напряжения отключения. <DC 10V> <S 3.3 V на батарею>",
" Температура Сна <С>", " Переход в режим Сна <Минуты>",
" Переходит в режим ожидания <Минуты>",
" Акселерометр <0. Выкл. 1. мин. чувствительный 9. макс. чувствительный>",
" В чем измерять температуру", " Шаг измерения температуры",
" Скорость обновления дисплея температуры",
" Ориентация Дисплея <A. Автоматический L. Левая Рука R. Правая Рука>",
" Активация кнопки A для Турбо режима до 450С при пайке ",
" Установка температуры для Турбо режима",
" Изменяет стрелки на дисплей питания при пайке",
" Автоматический запуск паяльника при включении питания. T=Нагрев, S=
Режим Сна,F=Выкл.",
" Мигает температура на экране охлаждения, пока жало остается горячим." };

const char* TempCalStatus[3] = { "Cal Temp", "Cal OK ", "Cal Fail" }; //All fixed 8 chars
const char* UVLOWarningString = "Low Volt"; //Fixed width 8 chars
const char* CoolingPromptString = "Выкл. "; //Fixed width 5 chars
const char SettingTrueChar = 'T';
const char SettingFalseChar = 'Ф';
const char SettingSleepChar = 'С';
const char SettingFastChar = 'Ф';
const char SettingMediumChar = 'M';
const char SettingSlowChar = 'С';
const char SettingRightChar = 'Р';
const char SettingLeftChar = 'Л';
const char SettingAutoChar = 'A';
const char SettingTempCChar = 'C';
const char SettingTempFChar = 'Ф';

#endif

runy1984, 2017-09-17:
what else to do?

runy1984, 2017-09-17:
as an option [http://arduino-project.net/russkie-ukrainskie-shrifty-oled-i2c/](url)

runy1984, 2017-09-17:
http://arduino-project.net/russkie-ukrainskie-shrifty-oled-i2c/

Ralim, 2017-09-17:
@runy1984 
If you can source a 12x16 font then I can use that to create the characters for the font map.
However, the font must also have a licence of some form to go with it, as I don't want to be guessing and getting it wrong. 😄 

Once I make up the font table for it (or you can make one like in issue 87 and save the code for each character) then I can add support for it and then make the build in Russian 😁

runy1984, 2017-09-17:
[OLEDI2C.zip](https://github.com/Ralim/ts100/files/1308918/OLEDI2C.zip) ок ?

Ralim, 2017-09-17:
The code only includes a 6-pixel high font by the looks of it. This should be workable in the 2.x version for the small font. However, a large font will still need to be made (tiny fonts do not scale well sadly).

I will test porting the small font over however :)
It does look like the licence is supportable too which is good.

runy1984, 2017-09-17:
when can you do the translation???

runy1984, 2017-09-18:
when you can do it in Russian?

Ralim, 2017-09-19:
@runy1984 
I need to create at a minimum a 12x16 font for the LCD which I have not been able to find online anywhere.
Making a font takes time and I mostly only have time to spend on this project during weekends at this point in time.

It will take some time to create the font from scratch (especially as I do not know the Cyrillic alphabet).

runy1984, 2017-09-19:
OK! thank you very much for the help!!! they're Russian firmware!!! very much we wait!!!

uncle-fed, 2017-09-25:
I am working on creating the font already, as I mentioned here
https://github.com/Ralim/ts100/issues/87#issuecomment-328835815

Shall I stop? I don't want to do a lot work requiring precision and time, if somebody is doing already.

runy1984, 2017-09-25:
do we all look forward to!!!

uncle-fed, 2017-09-25:
Then I suggest not to create more issues for the same topic but to keep it in one.
I created issue #87 already for Cyrillic font / Russian / Ukrainian translation, there is no need to have two issues for this. Can we please move this discussion into the original issue #87 and close this one?

runy1984, 2017-09-25:
Yes as you see fit and do it!

Ralim, 2017-09-26:
Closing this issue as I agree with @uncle-fed that their issue is the main root for this one.

closed 2017-09-26 by Ralim