Ralim/IronOS #95: [v2.0 FR] Chinese/Japanese/Korean translation

dogtopus, 2017-09-20:
IMO there are several things that need to be done to make this possible:
1. Fonts - we can use [GNU Unifont](https://en.wikipedia.org/wiki/GNU_Unifont) for this (in the upcoming v2.0 licensing should not be an issue since v2.0 will use GPLv3 and Unifont is GPLv2). The same font could also be used on all other languages (including English) due to its large coverage of characters.
2. Update the font handling code so that it supports 8x16 and 16x16 font dimension (if using Unifont) and sparse glyph table (i.e. only compile the used glyphs into the firmware, or otherwise the font data won't fit into the STM32's flash)
3. And of course, the text for each languages (I can contribute a Chinese translation after all the needed feature were implemented)

Ralim, 2017-09-21:
Hi,
The Unifont is a really good suggestion as a font to use!
1.
However, at 16x16 I feel that 6 characters really will not leave much room on the screen for the various options (though I have very little idea about other languages (sorry), so this could be wrong).

Also for 2.x and onwards there is the half height font (maximum of 8 pixels tall) that is used for the detailed screens for information. Would this be feasible?

2. The font code for 2.0 itself should be okay to scale for the font itself so long as its height is 8 or 16 pixels. (Don't support un-aligned writes to the screen at the moment). Sparse Glyphs would be a very good idea to add to the project, However, at the moment we do have room to spare for encoding the Glyphs.

3. 😄 Hopefully this is the easiest part

perillamint, 2017-09-21:
Korean here. IMO, In case of Hangul, we could use "Dokkaebi fonts" which composite Hangul glyphs on-the-fly using Hangul Jamo chunks.

By using this method, font ROM will take about 12KiB, much smaller than several megabytes of font image.

Here is my humble documentation about that method: https://wiki.gentoo.moe/wiki/Dokkaebi-decomposed-font

perillamint, 2017-09-21:
@Ralim Chinese (and its derivative alphabets) and Korean alphabets tend to much more "dense" compared then latin alphabets, so it could be a lot tricker task, but not impossible.

For example, "Low volt" message could translate into "전압 부족" in Korean and "低電壓" in Hanja (sorry, it isn't "Chinese version" of Hanja, I just typed "Korean version" Hanja), which will fit in OLED screen.

Ralim, 2017-09-21:
@perillamint 
12KiB could work okay (there is the room) for the fonts, is that including both a 16-bit tall and an 8-bit tall glyph set?

I'm aware that they are much denser than latin alphabets, just not sure *how* dense they are, and how well they handle being shrunk to 8-pixel tall characters?

perillamint, 2017-09-21:
Maybe Hangul *can* survive under 8x8, but it will hurt readability a lot.

Chinese Traditional glyphs will not survive under 8x8 due to its complexity.

I don't have any experience about Chinese Simplified, so I couldn't tell about it.

perillamint, 2017-09-27:
FYI: Here is data from Twitter, which analyzed tweet length difference between CJK and non-CJK.

It tends CJK alphabets are approx 2x denser(in meaning of information Entropy per char) than latin alphabets. https://blog.twitter.com/official/en_us/topics/product/2017/Giving-you-more-characters-to-express-yourself.html

Ralim, 2017-09-28:
@perillamint 
The 8x8 mode is just for the advanced screens, so it might work to have it fairly low readability since it only shows 3 values (We could also do pictoral symbols instead of actual words if that works easier ?)

dogtopus, 2017-10-01:
The smallest pixel simplified/traditional Chinese font I ever saw was 12x12. The readability was OK but a lot of details are already missing and certain characters look weird (although the developer of that font already tried their best to optimize it). So I would say 12x12 is probably the smallest you can get for Chinese (both simplified and traditional) characters (except for the really simple one)

Ralim, 2017-10-02:
@dogtopus Thankyou for this.

At this point, I'm thinking that the best way to go from here would be to support these glyphs in the 16 high larger font space, but if the user drops to the advanced screens these will have to be in Latin characters just because of the low resolution used to show more information.

I can add support for 16x16 fonts if this will be the easiest, however, I do not currently have the time to convert to the bitmapped font table just yet. I can get to this later on though 😄 

Would it be better to look at a 12 wide font, or add the font as 16 wide? (Doesn't really matter to me, just not sure if the extra density of 12 would be worth it?)

dogtopus, 2017-10-03:
@Ralim I would say that the safest way is to include the 16x16 font. The [12x12 font](https://github.com/SolidZORO/zpix-pixel-font) I saw uses a custom license (looks like CC-BY-NC-ND+paid commercial license according to the description from an older version (current version doesn't even have a clear license since the author removed the old license terms)) which is incompatible with GPLv3. Also I am not sure how recognizable the glyphs of Unifont will be if they are being scaled and rendered at 12x12 resolution.

Ralim, 2017-10-04:
@dogtopus 
Fair enough :)

I'll try and get Unifont converted over soon. (Might be a delay here, down a pc at home atm)

Ralim, 2017-12-12:
I've been way to busy to convert the font at the moment.
Going to let issue #149 track this conversion.

closed 2017-12-12 by Ralim