Ralim/IronOS #96: Danish Translation

t-xdk, 2017-09-25:
it would be nice to have it translated into danish

```
#ifdef LANG_DK
  const char* SettingsLongNames[14] = {
    /*These are all the help text for all the settings.*/
    /*All must start with 6 spaces so they come on screen nicely.*/
    "      Strømforsyning. Indstil Cutoff Spændingen. <DC=10V S=3.3V per cell>",
    "      Dvale Temperatur <C>",
    "      Dvale Timeout <Minutter>",
    "      sluknings Timeout <Minutter>",
    "      Bevægelsesfølsomhed <0=Slukket 1=Mindst følsom 9=Mest følsom>",
    "      Temperatur Enhed <C=Celsius F=Farnheit>",
    "      Temperatur Afrunding <Units>",
    "      Skærmtemperatur Updaterings Frekvens <L=Langsom M=Medium H=Hurtigt>",
    "      Skærm Orientering <A=Automatisk V=Venstre Håndet H=Højre Håndet>",
    "      Ved tryk på front knap Aktiveres boost-funktionen, 450C tilstand når der loddes <j=Ja N=Nej>",
    "      Temperatur i \"boost-funktionen\" <C>",
    "      Ændrer pilene til et strømdisplay ved lodning <j=Ja N=Nej>",
    "      Start automatisk med lodning når strøm sættes til. <L=lodning D=Dvale tilstand S=Slukket>",
    "      Blink temperaturen på skærmen, mens spidsen stadig er varm. <j=Ja N=Nej>"
  };
  const char* TempCalStatus[3] = { "Kal. Temp", "Kal. OK  ", "Kal. Fejl" }; //All fixed 8 chars
  const char* UVLOWarningString = "Lav Volt"; //Fixed width 8 chars
  const char* CoolingPromptString = "Sluk  "; //Fixed width 5 chars
  const char SettingTrueChar = 'J';
  const char SettingFalseChar = 'N';
  const char SettingSleepChar = 'D';
  const char SettingFastChar = 'H';
  const char SettingMediumChar = 'M';
  const char SettingSlowChar = 'L';
  const char SettingRightChar = 'H';
  const char SettingLeftChar = 'V';
  const char SettingAutoChar = 'A';
  const char SettingTempCChar = 'C';
  const char SettingTempFChar = 'F';
#endif
```

Ralim, 2017-10-01:
Hi,
I haven't missed this or forgotten about it! 
Thank you for contributing this as well :)
I'm just currently getting translations working in 2.x at the moment so it will be up when the other languages come up in 2.x (soon hopefully, time permitting).

Ralim, 2017-12-12:
@t-xdk Can you update this to the new 2.x format And I'll try and pull this in and get it working.

t-xdk, 2017-12-12:
Danish V2 translation

> #ifdef LANG_DK
> const char* SettingsLongNames[16] =
> 		{
> 				/*These are all the help text for all the settings.*/
> 				/*No requirements on spacing or length*/
> 				"Strømforsyning. Indstil Cutoff Spændingen. <DC 10V> <S 3.3V per cell>", //Power Source
> 				"Dvale Temperatur <C>",    //Sleep Temp
> 				"Dvale Timeout <Minutter/Sekunder>",    //Sleep Timeout
> 				"sluknings Timeout <Minutter>",    //Shutdown Time
> 				"Bevægelsesfølsomhed <0.Slukket 1.Mindst følsom 9.Mest følsom>", //Motion Sensitivity
> 				"Temperatur Enhed <C=Celsius F=Fahrenheit>",    //Temp Unit
> 				"Vis detialieret information med en mindre skriftstørrelse på standby skærmen.", //Detailed Information
> 				"Skærm Orientering <A. Automatisk V. Venstre Håndet H. Højre Håndet>", //Orientation
> 				"Ved tryk på front knap Aktiveres boost-funktionen, 450C tilstand når der loddes", //Boost enable
> 				"Temperatur i \"boost\" mode",    //Boost Temp
> 				"Start automatisk med lodning når strøm sættes til. L=Lodning, D= Dvale tilstand,S=Slukket", //Auto start
> 				"Blink temperaturen på skærmen, mens spidsen stadig er varm.", //Cooling Blink
> 				"kalibrere spids temperatur.",    //Calibrate Tip
> 				"Gendan alle indstillinger",    //Reset Settings
> 				"VIN kalibrering. Knapperne justere, Lang tryk for at gå ud", //VIN Cal
> 				"Vis detialieret information mens der loddes",    //ADV SLD
> 		};
> 
> const char* SettingsCalibrationWarning =
> 		"Sørg for at loddespidsen er ved stuetemperatur, inden du fortsætter!";
> const char* UVLOWarningString = "Lav Volt";    //Fixed width 8 chars
> const char* SleepingSimpleString = "Zzzz";    // Must be <= 4 chars
> const char* SleepingAdvancedString = "Dvale...";    // <=17 chars
> const char* WarningSimpleString = "Varm!";    //Must be <= 4 chars
> const char* WarningAdvancedString = "ADVARSEL! VARM LODDESPIDS!";
> 
> const char SettingTrueChar = 'j';
> const char SettingFalseChar = 'N';
> const char SettingRightChar = 'H';
> const char SettingLeftChar = 'V';
> const char SettingAutoChar = 'A';
> #endif

commit 87c2ed4f (2017-12-12): Add Danish

Closes #96
closed 2017-12-12 by Ralim