Ralim/IronOS #97: Just found this ts100 as an Oscilloscope

xavkno, 2017-09-26:
Aperrently this is based on this firmware: https://youtu.be/Gg9mKVMbp0I

mikromCZ, 2017-09-26:
awesome! I found Tetris instead: https://www.youtube.com/watch?v=Buzew1z1AhQ

vifino, 2017-10-03:
I'd love to see this integrated into the main firmware. Just get a common ground and you're ready to party.

Ralim, 2017-10-03:
@vifino This modification requires a manual hardware mod to wire the ground of your tip through to one of the ADC channels (they use one of the buttons). This means if you poke your iron at any voltage above approx 3.5V relative to your iron power's ground you will damage that input and not be able to use your iron correctly again. This also opens up ESD damage to the ic and prevents you from having your tip grounded, having a grounded tip is usually a requirement for most uses of soldering irons.

I'm hesitant to include this sort of feature because its opening up a whole world of ways people can break their units (Remember this also opens the unit to ESD damage through the tip).

The Tetris one is known about, and I've thought of including it, but it really just comes across as a playtoy  that really isn't needed on a *tool*

vifino, 2017-10-03:
@Ralim Oh, yuck. Nevermind then! I'd rather have a working iron than a broken iron that could pretend it's an oscilloscope for all of 5 seconds. :)

mikromCZ, 2017-10-05:
oh! they made also screwdriver! opportunity for another custom firmware?
https://www.youtube.com/watch?v=il86j54dTzc

Ralim, 2017-10-05:
Yep I know the ES120 rather well.
Already started on an alternate firmware for it :) But its probably at least 50 hours of dev time away from being useable as a firmware sadly. (Takes a fair bit to get everything up and running)

perillamint, 2017-10-10:
@Ralim BTW, did you managed to connect SWD to ES120? I managed to disassemble it (thanks for disassembly tips :)). Like what you said, there are no exposed test pads waiting for us. I think the only way to connect SWD is solder directly on STM32F103, but it requires very fine iron tip and a quite skilful touch (aka, microsoldering) which I don't have both. :(

Ralim, 2017-10-10:
@perillamint I didn't try for SWD and instead just developing debuggerless atm :)
The original firmware & bootloader turn off the SWD by the looks of things too.

EricJanVanDenBogaard, 2017-10-30:
The TS100 is great as a soldering iron but not as oscilloscope or games machine i guess :)
Maybe if no firmware updates are needed, and there is still any memory available then....
I think the TS100 is a great hardware tool and we are blessed with some great people able to make max use of it.

SijmenSchoon, 2017-11-01:
It's a really cool idea, but not really that feasible. Let's not let feature bloat get to this feature — as the Unix philosophy says: "Make each program do one thing well. To do a new job, build afresh rather than complicate old programs by adding new 'features'."

joric, 2017-11-19:
@mikromCZ they've made tetris on the es120 screwdriver as well :)
https://www.youtube.com/watch?v=Q2bpTqUKbqA

closed 2018-04-27 by Ralim