Ralim/IronOS #99: Feature Request: adjustable time for logo

michael-blue2, 2017-09-27:
Is it really hard to make 1 more setting menu which will let you select how many seconds the logo will appear? :D

Ralim, 2017-09-28:
On the list of things to change is to make it just dismiss with button press 😄 
Is that okay ?

Ralim, 2017-09-28:
Okay, for now, I've made it wait 1 second or skip on button press. Thoughts ?

michael-blue2, 2017-09-28:
Isnt it the default that comes with 1s? I think 2seconds will be perfect woth the skip aswel on button press:) sometimes you just want to show that logo around and the other times you just need to solder and dont care about it :)

Ralim, 2017-09-28:
Sounds fair, I can bump it up to 2 seconds easily.

michael-blue2, 2017-09-28:
perfect then :D, should i close the issue/request?

tigert, 2017-09-28:
The show-until-buttonpress would be problematic because the idle image after the logo shows the hint for what button turns the iron on

Ralim, 2017-09-28:
Show-until-buttonpress shows until either button is pressed and also 'consumes' that button press so it should not cause anything else to occur after its dismissed.

closed 2017-12-12 by Ralim