Ralim/IronOS #101: Feature request: make boost mode latching

venumz, 2017-09-29:
First of all, awesome work, your firmware is brilliant.
Only thing I find annoying is that the boost mode requires holding down the button the whole time you need it up. Maybe you could change the state to boost mode as soon as it starts and leave it there until you press the button (nearest to the tip).
Regards

tigert, 2017-09-29:
That is kinda the point of the boost mode.

If you need to solder something bigger and think you need more heat, may I kindly suggest you to try the C4 tip?

venumz, 2017-09-29:
Thanks for the suggestion, my tip is fine, is just that I lose precision when I'm holding down the button so a latching option would be awesome for me.

Cpt-Huck, 2017-10-02:
If you loose precision, why dont you just change the temp to a higher one instead ? In my opinion, I would forget to disable boost mode. Holding the button is fine for me.. idk.. but this should not be implemented without an option to disable that !

Ralim, 2017-10-02:
Unfortunately, I'm with @tigert and @Cpt-Huck here.
If you are using boost mode all the time, why not just adjust up the soldering temperature?
Its intended to only be used when you're about to do a small number of joints that are larger than normal.

One thing to note, if you are instead doing a lot of large joints with a small few joints, you can actually use the boost mode in reverse by setting the temperature to *lower* than your normal temperature, so when you hold the button the tip temperature will actually drop. I have not found any real use for this, but there is nothing stopping you from doing this if desired.

It would be possible to add the option to make the boost mode latching, however, I feel that its kind of against its use case?

venumz, 2017-10-03:
Thanks for your feedback, it was just an idea, I can live without it.
I use to do smd work mostly, but in my boards sometimes I have big through hole joints, I use boost mode there, I figure that I can change the grip a bit to make it more comfortable holding down the button.
Your firmware is awesome anyway, keep up the good work.

tigert, 2017-10-03:
Was there an initial idea that the boost mode goes higher than the otherwise maximum temperature, and thus would only be maintained while actively pressing the button, to avoid damage to the device if it was accidentally left on for a long time? I am not sure if this is actually the case though.

Ralim, 2017-10-04:
@tigert Sort of...
The original firmware only allowed you to set the tip up to 400C, whereas in this firmware I have raised that to 450C. The idea of the boost mode was to try to encourage users to solder at the ~350C or so normally and only boost up when needed for larger connections to try and help tip life and stress on the unit itself (as the handle starts to get quite warm at the higher temperatures).

I never really enforced this as it seemed pointless and easier to let people decide for themselves (especially as the tips are cheap).

Mrkvozrout, 2017-11-22:
I think that holding the button is just fine as it is not that stiff to influence handling precision.

But just as a thinking exercise:
The request could be implemented as a "Deadman's switch" similar to its usage in trains (see https://en.wikipedia.org/wiki/Sifa). The train driver needs to press a pushbutton regularly to keep going - if driver stops to press the button (e.g. fainted), the train will automatically stop.

So there could be an option in settings that will hold number of seconds to keep boosting after the boost button is pressed.

- 0 would be for current state (boosting while pushing) 
- positive int (up to some limit) would be number of seconds to boost when button is pressed
- e.g. configured 5s - boost pressed > boosting for 5s, then return to target temp
- boot pressed (boosting for 5s) but boost button pressed again after 3s > boost from now for 5s
- holding button could behave same as currently with boosting stop after button release (button release should cancel any ongoing "boost latch")

SmayK, 2018-03-16:
I think we have a good idea.
the 5s hold activates the boost mode
in the menu setting boost, for this maintenance have the possibility to choose the time from 3s to 10s
example I set it in "Boost Time" to "4s"
I press the button 5s, the boost mode is activated and displays "Boost Mode" with a countdown from 4s to 1s and returns to the normal heating display.

discip, 2022-04-23:
@venumz
As pointed out by @Ralim and agreed upon by yourself, this is kind of pointless.
So please close this issue. 😊

thanks

venumz, 2022-04-26:
Closing the request!
Thanks for your amazing work!

closed 2022-04-26 by venumz