Ralim/IronOS #102: while soldering display if element is heating or not

vince-br-549, 2017-09-30:
While running with an input of 12 volts and attempting to solder a large item the  temperature  doesn't  seem to change yet the  iron is not able to melt the solder.

I expected to see two things.  1. the temperature to drop, and 2 some indication that the iron was attempting to apply full power  to the heating element.

So I don't know if the power to the heating element is digital ( on or off  only ) or variable.
If it is digital, then I would like the display to show me some kind of ON or OFF indication, rather than the three horizontal bars  that is does currently.  If it is an analog or variable switch then maybe it should show say 1, 2, or three dots as an indication of the  amount of power being applied.

aside comment,  The  DC5525 connectors I found are  horrible the make very poor contact.  Looking for better ones

Ralim, 2017-10-01:
Hi,
Can I first ask which firmware you are running? 1.x or 2.x series?
If you are running the 1.x series can you possibly try the 2.x series as it has a newer, better method for the tip control?

When you are running at 12V there isn't a huge amount of heat present at the tip (~35W), so it will take time for the other part of the joint to heat up.

The tip control is variable (PWM), and there have been experiments with trying to show the heating amount being applied but this quickly runs into an issue as when running at 12V it will quickly be ramping up to full power to regulate the temperature, however with ~20V input that most people use it will rarely ramp up to the full 100% power. So the scaling on the indication is related to both the input voltage and also the tip that is being used as some (such as the small C tips) have wildly different thermal mass.

There is also a thermal delay in the tip between the outside temperature dropping and the internal heating element temperature dropping (temperature is sensed at the heating coil).

Please also ensure you have a solid contact between the tip and the join you are trying to heat (add flux?), I know you probably know this but doesn't hurt to ask :)

vince-br-549, 2017-10-01:
I am running V1.17.1 .  I just now see the V2.
PWM  ahh that is good.  I just converted from 12V to 20V  maybe I'll wait a
little on the V2 series and see how it goes.
I'm using the D24 wedge tip.

I guess I have to order me another DIY soldering kit to get a feel for this
new toy.

By the way  upgrading via Linux was trivial.  I didn't use the script.

Thanks for your quick response.  I just now turned on watch for this
project :)

---
Vince

Ralim, 2017-10-01:
I would suggest trying the 2.x firmware just to see. It should perform massively better at 20V than 12V.
Yeah, its PWM but its a double PWM implementation so it can only update the PID at ~10Hz which is not ideal but its about the best that can be done at the moment. (Still toying with it).

The D24 is a fairly good tip for larger work and should be working fine for your joints. Maybe just try adding a little extra solder to bridge the gap when it seems to not be working overly well.

Linux is *mostly* easy :)

No problem, I *try* to get back to things quickly 😄

tigert, 2017-10-01:
if you need more heat, the C4 tip (the large, round, angled) has been great for me for bigger things like soldering thicker battery wires to solder pads. 

If you are new to soldering in general, it is also about technique. It helps to pre-tin the wires and pads with solder first, possibly adding flux, and then just heating the thing to melt it together.

dogtopus, 2017-10-08:
There is an optional tip power indication in v1.x that may partially fulfill what you want, but disappeared on v2.x
Kinda want it back :(

Ralim, 2017-10-09:
@dogtopus The current 2.x firmware is set up so that the heating icon appears when the PWM is > 10%.
In the 1.x firmware, the bar graph showed a scale of roughly 0-20%.
Generally, if you want to see if the tip is performing significant (ie not just temperature maintenance) the icon will change to heating.

I didn't add it back in 2.x because it was never really calibrated and never received any feedback when trying to calibrate it and never saw any use of it mentioned and no one I asked used it.

closed 2018-03-03 by Ralim