Ralim/IronOS #104: Two minor suggestions (text related)

rawrsy, 2017-10-04:
Hi Ralim, love the firmware. Running 2.0 but I would like to make two little suggestions that *may* or may not make the thing easier to use. The first, is to replace the "WARN!<temp>" message with "HOT!<temp> might make it look a bit neater.

Second suggestion, came to light when I noticed how readable the little version and date bootup message is with the tiny text. Would it not be a good idea to render the menu items in that small text, to fit more info on the screen and make the options a bit less confusing for a dill like me?

Those are just my 2 cents, please feel free to take them for what they're worth (about 2 cents).

Many thanks for reading, and for your amazing work.

Rory

Cpt-Huck, 2017-10-04:
Both suggestions are great!

Cpt-Huck, 2017-10-04:
Also testing the newest FW atm and will report bugs if I find some!

Ralim, 2017-10-04:
Hi,
The text "WARN" has also been ("COOL","HOT","!!!!") in previous firmware, and every time people ask for it to be changed someone else wants it to be something else :grimacing: 
As this is the 2.x alphas I want to hear this feedback! 
I might swap it for "HOT" in the next alpha and see how it looks.

For the second suggestion, the big reason this cannot be done is because of the multiple language support sadly.
English's Latin alphabet is one of the very few that can work at low resolution, and as the settings menu is the same code for all languages we cannot use anything that would not work with other languages.
Trying to shrink some languages glyph set down to 8 pixels tall is almost impossible sadly.

So the choice was made to keep the settings menu with the large font (with help text) so that the unit could support more than just English.

Ralim, 2017-10-06:
Alpha 3 should have the text swapped to "HOT!"
Nothing I can really do for a small font unfortunately :(

closed 2017-10-28 by Ralim