Ralim/IronOS #105: Q - Do we need boost mode enable setting ??

Ralim, 2017-10-04:
Hi,
It seems like everyone who comments here seems to use the boost mode these days.
Does anyone like to have it turned off for any reason?

Just thinking it could clean up and remove an extra menu item if no-ones really changing it much.

thanks4opensource, 2017-10-05:
Maybe I'm the only one, but I do **not** use boost mode. When I need to change temperature it's simple enough just to use the "up"/"down", "tip"/"LCD" buttons.

I'll have to check, but how does real/normal temperature change work in boost mode? Do you have to press the LCD button for a mode change so the buttons raise and lower the temp setting? If so, adds another step for us "non-boosters". Also runs risk of accidentally long-pressing the LCD button (BTW still an excellent user-interface option vs. two-button) and going into shutdown mode.

My vote: Please keep non-boost mode in. Would be fine with me if "BOOST T" was the default.

Ralim, 2017-10-05:
So how it works is : Whilever you hold the front button it will change the setpoint temperature its set to, to the boost temp instead. When you let go of the button it will drop back down again. Short pressing the buttons will still just enter temperature changing as per normal.
Long pressing the usb end button still just exits as per normal.

Thanks for your vote though :)

thanks4opensource, 2017-10-05:
Thanks, @Ralim. Sounds like boost mode doesn't get in the way of normal temperature setting, so I won't miss turning it off too badly if it goes away. Still think the more customization options and choices the better. :)

Trezona, 2017-10-05:
I love the boost mode. I use it quite often. I find it very handy.

Ralim, 2017-10-06:
I'm definitely leaving it in, was just wondering if I should leave the menu option. Which I will for now :)

closed 2017-10-06 by Ralim