Ralim/IronOS #106: sleep mode and auto shutdown won't work

lnnocentius, 2017-10-06:
For some reason, any build beyond V2.0 of the Ralim firmware does not activate it's auto sleep feature. But on V1.17 and even on stock firmware everything works fine and dandy, so I think we can rule out that my unit is wonky.

My settings for my iron are:

350 degC temperature
PWRSCR DC
STMP 150
STME 10S
SHTME 05
MSENSE 8
TMPUNT C
ADVDSP T
DSPROT A
BOOST T
BTMP 450
ASTART F
CLBLNK F
also calibrated my VIN at +0.2 volts approx

mine is the orange TS100 version if that helps
I've also reflashed the latest V2.x firmware to rule out a faulty copy of the firmware but no dice.
Thanks for all the hard work you have put on this project

Ralim, 2017-10-06:
Hi,
In the new 2.x firmware the motion sensitivity is alot more sensitive. Depending on the particular tolerance of the ic used in your iron it may not be able to sleep at a motion sensitivity of 8.

Could you try taking the sensitivity down to say 5 testing there and bringing it back up?

thanks4opensource, 2017-10-06:
@lnnocentius -- what Ralim said. MSENSE is much improved in the 2.0 alphas. Turn it down to your liking. For me, 8 or even 7 stays awake in my hand but goes to sleep when set down (I use "STME 10S" like you). It's perfect! ;)

@Ralim -- 2.0 alpha3 continues the improvements. Thanks! Degrees F everywhere I've checked for it, and "HOT!" changes to start menu (@tigert, the new icons look great) and then LCD off at 86F instead of waiting for 30 or 50F which never comes.

But still a problem with "HOT!" vs shutdown: I think it's the opposite of alpha2, but now only works correctly ("HOT!" until 86F) if manually shut down via long LCD/cord-end button press. If timeout automatic shutdown from sleep state, LCD immediately blanks and button press gets startup icons -- no "HOT!" warning at all. Could you test with the following settings, including "SHTME 01" so you don't have to wait a long time to see it happen, and see if it's just my TS100? Thanks.

```
PWRSC DC
STMP 50
STME 10S
SHTME 01
MSENSE 8
TMPUNT C
ADVDSP F
DSPROT R
BOOST F
BTMP 420
ASTART F
CLBLNK T
TMP CAL?
RESET F?
CAL VIN?
```

BTW, "ZZzz" on sleep: Very cute. :)

Ralim, 2017-10-07:
@thanks4opensource 
I could replicate that locally. Looks like its patched in the Alpha 4 if you could try :)

thanks4opensource, 2017-10-07:
@Ralim -- Two alpha releases in one day! It's like having my own personal software developer! ;)

Shutdown warning screen works for me, both via timeout and explicit button press. Fixed/good/thanks. A few more minor comments ...

Maybe one more pass keeping everything in internal sensor temperature units and converting to degrees C or F only when displaying or getting input? Default temperature on fresh install is 320(??) and stays at that when changing to F, which is then very low. Not a big problem, don't think people will be switching back and forth between F and C that often. A little worse going from C to F -- set to 550F, change to C, way too hot next time heater is turned on.

More problematic with boost temperature. Tried it, and default 420 in "F" is "negative" boost (as you documented works, but kind of useless because cool-down is so slow). Again, the display/setting of BTMP in setup menu is currently dimensionless -- if stored internally in native units and converted for display it would even track dynamically in setup as F and C were toggled.

Very small/debatable "ask": I wish regular (non-menu, non-boost) temperature would increment continuously when buttons held down. Again, need this only rarely (like when going from C to F and having to do large changes) and most users probably do only small temp changes -- plus use boost for big ones. Still, works that way in the menu system, and as long as the delay didn't make it touchy to only do single increment/decrements, I think it would be helpful.

Finally, one tiny one: Love the DC power plug icon, but it should be on the cord end of the display in all orientations ("A" depending on how held, and "R" or "L" if explicit). BTW, also like the international triangle-with-!-sign for shutdown temp warning vs. all the "WARN", "HOT", "COOLING" alternatives.

As always, appreciate all your work. Become a commercial, closed-source developer and then you can ignore all your user's requests! ;)

lnnocentius, 2017-10-07:
@Ralim --- tried to turn down MSENSE to 5, clamped my ts100 to my bench vise to be sure it won.t move. Still no dice I'm afraid. I checked the seating of the STM 32 module but still no difference. Maybe a bug on how low I set my SLTMP? I'll try setting MSENSE to 3 later when I get home. It's also the same with shut down time

lnnocentius, 2017-10-07:
Also, I don't know if this will mean anything, but when my iron is idle and I have set it to a temperature of let's say, 350 degC, it will continuously show a temp that is floating at 348-352. It never stays still on 350 degC.

Ralim, 2017-10-07:
@thanks4opensource 
Well its different days to me :P (Aussie)
It's a bit of a pain to store the thresholds in the internal raw readings as those are not compensated for the cold junction temperature (actual temp is raw + cold junction temp - calibration value). 

Moving it to match the cord makes sense, I'll look into that one though :)

Haha, I enjoy the open source community since I can have the community engagement like this :)

@lnnocentius 
Okay that's sounding rather odd.
The big change is that in 1.x I used the accelerometers internal motion detection system. Which while okay, was not sensitive enough for some users. So in 2.x It's reading the raw readings and calculating the deviation per 300mS. (which is compared to the threshold).

I sensitivity of 0 disables sleep and shut down entirely, but a sensitivity of 1 would be the least sensitive.
Ill try and get a debug build tomorrow for you :)

lnnocentius, 2017-10-07:
tried MSENSE 1, still not going into sleep mode nor shutdown. Maybe my TS100 is secretly consuming my coffee stash

Ralim, 2017-10-09:
Hmm damn okay, I'll build up a debug firmware that prints out the readings from your accelerometer for diagnostics.

Ralim, 2017-10-18:
Hi @lnnocentius,
Could you try the attached firmware? It will print out the LCD the readings from your units accelerometer to see if it's just not reading correctly or if there is something funky with one of the axis.
It should pause on startup and then show 3 numbers if you leave it still these should settle down to fairly low values (<100). Let me know what happens on your unit :)

[TS100.zip](https://github.com/Ralim/ts100/files/1393892/TS100.zip)

lnnocentius, 2017-10-21:
@Ralim All of the axis that is displayed is spiking about a little above 100. The top right one is the jumpiest. The TS100 was resting on my soldering stand with this reading.  The iron was resting at a 30 degree angle.

When I clamp the iron horizontally to my bench vise, the bottom left readout is the only one that is going above 100 (just frequent spikes towards 100, usually rests at 32-16)

Here is a link on how it behaves
https://youtu.be/KchZQi-fTHo
I'll try out the new build if anything changes

Thanks again for helping out on this problem

78suited, 2017-10-23:
Same problem here. When using the default FW from MiniWare it works fine (v2.17) and also with the Ralim V1.17.1 release. But with the latest Alpha V2.0 Alpha 5 #2 it doesn't. Have been playing around with all settings but still no sleepmode or auto shutdown.

LexioTech, 2017-10-26:
Same issues here, my TS100 arrived today, tried it with the stock firmware, the sleep mode works, flahed the latest 2. something 5 build no sleep, no auto off :) I hope this can be fixed because otherwise I love the UI and the settings are so nice :) Keep up the great work!

Ralim, 2017-10-26:
Hi,
Still trying to figure out why your irons, in particular, are having issues.
The numbers @lnnocentius has in the video are perfectly normal.
The system shouldn't trigger movement until the summation of those numbers is > 600 at max sensitivity, and > ~2000 at minimum sensitivity.

I have attached another tester firmware, that will print the value its testing against in the bottom right, If someone could try this one and just note down their rough numbers it would help me trying to find the issue :)

[TS100.zip](https://github.com/Ralim/ts100/files/1420258/TS100.zip)

78suited, 2017-10-27:
@Ralim https://youtu.be/Vsnm2wsSUEs is this okay or do you want to see something else?

lnnocentius, 2017-10-27:
@Ralim here you go, the bottom right is just spiking from 0->100
https://youtu.be/vVYVJcKl0xU

Ralim, 2017-10-28:
Hey,
Thank you for your videos, they do show that the readings are mostly updating well, though the slight pausing gave me a hint to what it could have been, as my irons do not do that.

Both your irons readings are very close to what my one here is reading as otherwise.

Could you try this build, I've fixed a few small bugs, which may have caused more issues on your irons than on mine and exasperated the problem.
[TS100.zip](https://github.com/Ralim/ts100/files/1423860/TS100.zip)

78suited, 2017-10-28:
@Ralim works like a charm, did a little test and both functions work now. Much appreciated, keep up the good work! https://youtu.be/vkuQ9R_4jvc

christer, 2017-10-28:
@Ralim Also tested and it works on mine too.

LexioTech, 2017-10-28:
Awesome, thank you 👍 Works perfectly :)

LexioTech, 2017-10-28:
One little thing that (if possible) could be improved is the sensitivity. I'm all the way down to 1 now, and still if I bump my desk ever so slightly, the iron turns off the sleep mode and starts to heat up :) It works fine if you actually want to use sleep when you get up from your desk, but if you for example soldering for 3 minutes, then tinkering for 15 more, and then soldering again, the sleepmode will turn off each time you bump your desk. So if this can be improved, it would be 100% perfect firmware for me :) Thanks for the hard work and awesomeness!

lnnocentius, 2017-10-30:
@Ralim sleep now finally works, thanks for the hard work on fixing the problem

Also, I second what @LexioTech has commented. Even at MSENSE 1 the iron is very sensitive, as even simply nudging the iron a bit takes it out of sleep mode. I think a initial motion buffer can be added?

lnnocentius, 2017-11-08:
@Ralim the latest release fixed the sensitivity issues. I can now safely say that this thread is now resolved. Thanks for the hard work in fixing it

closed 2017-11-08 by lnnocentius

Ralim, 2018-08-20:
What PCB version is reported in the menu you get by holding the rear button on the unit when it's on the idle screen?

You can also use this screen to debug if yours is just too sensitive, there is a movement timer that it shows, this should not change while the unit is sitting idle. If it is your sensitivity is too high