Ralim/IronOS #107: Detailed view instead of Idle screen graphic (only)

csdpm, 2017-10-07:
The idle screen graphic is nice, but only really useful while getting used to the iron. It would be nice if there was an option to change this to the Detailed view screen (as happens when advanced display is enabled) - but for that screen ONLY.

I'm finding that the Detailed view is too small for my aging eyes for the Sleep and Shutdown modes (too hard for me to see the temperature) So I would prefer to keep those modes as the simple display.

Having the detailed view for idle only provides useful information (set temp etc) even if I have to squint a bit.

Thanks.

 P.S. I like the ZZzz, but it would be nicer as Zzzz or a slightly smaller icon. ;)

Ralim, 2017-10-16:
Hi,
This should be quite doable to implement for the next alpha of the firmware.
I can just split the setting into two, one for the idle screen and one for the soldering screens.

csdpm, 2017-10-18:
That's perfect thank you!

You even did the Zzzz! :)

closed 2017-10-18 by Ralim