Ralim/IronOS #108: Single press heating from Shutdown?

csdpm, 2017-10-07:
Sorry Ben, one more.

It would be good if you could turn the iron back on from Shutdown with a single button press. At the moment it needs two, one to go to the idle screen and then another to turn it on.

The situation I'm thinking is where you've just exceeded your normally adequate shutdown time threshold (far sooner than the end of cooldown return to idle) and you just want to pick up the iron click it on again.

With that, I think you will have achieved ergonomic perfection with Version 2. 🥇

Ralim, 2017-10-09:
No problem :)

I can make it so that if you press the front button on the shutdown (temp warning) screen it can jump straight to soldering mode rather than exit to the idle screen. 

The only gotcha with this is that the screen you are seeing for the warning is the same as the one that will appear on startup if the tip is hot etc (i.e. there is only one warning screen).
This means that pressing the front button on the temperature warning screen would also let you skip the idle screen and go straight to soldering.

I'm not sure how much worse this would be in terms of people accidentally starting soldering or not?

csdpm, 2017-10-10:
Thanks, that works for me as long as nobody else has an issue with it. :)

Hopefully it won't be a safety issue for anyone as, if powered up from cold, it's only one press to go to soldering from idle anyway. If the tip is already hot at startup, it implies that you've just been using it too.

closed 2017-10-18 by Ralim