Ralim/IronOS #109: Bad tip error

Kenneth-N, 2017-10-08:
While briefly playing with v2.0 alpha 4, I encountered error BAD TIP.
So I reset the settings and the error is not showing anymore.
Before the error I played around with lots of settings (sensitivity seems to work fine now) and possibly activated the temp cal while it was still +200 degrees. This was when I tried to get out of the menus by pushing both buttons at the same time, several attempts, and got into different menus by mistake. Maybe this fiddling caused the problem.
Still have problems pushing both buttons at the same time to exit menus.

dogtopus, 2017-10-08:
> and possibly activated the temp cal while it was still +200 degrees
That might be the cause, since the BAD TIP warning works by catching the abnormal high temperature reading on the tip IIRC, and if you calibrate the tip when it's still hot, it will add a large offset to the tip temperature reading, which may trip the BAD TIP warning.

Ralim, 2017-10-09:
Almost certainly the issue would have been the temperature calibration being done incorrectly.
As this would cause issues in the entire temperature decoding chain.
The reset settings would load a typical value for the temperature offset as well, so if it's performing well after the reset it's most likely the calibration.

Kenneth-N, 2017-10-09:
Is there any code preventing the calibration when the iron is way too hot?
If not, maybe it should be there.

Anyway, reported the issue just in case someone else have encountered it AND if I did not cause it by playing around with the settings :-)

So I will close this then.

closed 2017-10-09 by Kenneth-N

Ralim, 2017-10-09:
At the moment the system uses the brain of the operator for the calibration, as the iron handle *must* be at the same temperature as the tip. It's annoying to put lockouts on the tip in case the tip has a bad offset.