Ralim/IronOS #110: What will break first when overvolting?

mindfuucker, 2017-10-10:
Hello,

I have a 7S li-ion pack with a max voltage of 29.4V. This is 5V more then the recommended maximum voltage. If I replace the input capacitors, what will break next? Will the converter IC break? Wil the power dissipation melt the device?

Have any of you tried an over-voltage and dit yours live to tell the tale?

Ralim, 2017-10-12:
Hi,
I agree that the 150-ohm resistor that is used to limit the enable signal is the most likely candidate to fail.
At 29V you will also be overloading the input to the ADC of the stm32, which won't hurt it so to say, but it won't work effectively.

I would also be worried about the extra power dissipation in the MOSFET, it's not especially well heatsinked and so may heat up over user and be damaged as well (or at least make the handle uncomfortable).

I would seriously recommend that you do not go so far over the advertised limits of the device. The entire unit is designed for a specific purpose with designed in hardware abilities. 

Going needlessly over these limits seems like asking for trouble. Is that extra ~15% of power really required? You will probably hardly notice the difference in the performance for most tasks?

closed 2018-03-03 by Ralim

Clausinho, 2019-01-12:
Sorry for digging up this old issue.

I have overvolted my TS100 and am now trying to repair it with my dad. Could you specify which resistor this exactly is and where I would buy a replacement? Is it the row right next to the display and left to the power input? Would this be a suitable replacement? https://de.aliexpress.com/item/0603-Chip-Fixed-Resistor-SMD-Resistor-1-150R-100pcs-lot-free-shipping/32523900214.html

Sorry for my ignorance, I am studying computer science and don't have a big grasp of electrical engineering. My dad is an electrician, so hopefully I will be able to figure it out with him.
Would love a nudge in the right direction.

whitehoose, 2019-01-12:
There are only a couple of resistors on the board. Usually start at the connector and work each of the outer tracks first (in turn).  With power often the first components take the punishment but iCs are fragile too.
If you are fixing a circuit - you NEED either a meter or test lamp to prove continuity and give some comparison of expected and actual value.  Eyes are useful as a first test - look for evidence of burnt or melted areas, but there is no substitute for following continuity round the track. which requires a meter   Just because a component fails 99.9999% times doesn't mean you can take it for granted THIS time... a 1 second dab with a meter PROVES the point. 

Buying replacements can be easy - resistors are usually marked and there is usually enough left to measure the size so you don't order an 0603 when you need a 0805 (size matters).
Capacitors usually aren't well marked and the 6 and 8 legged beasties are sometimes purposely wiped of all identity ... and can be impossible to locate.

While you are learning to use a meter - learn to use Google - if you can identify ANY information, it's surprising what will (and won't) give results. ... then - again Google will find suppliers if your "usual" source doesn't list them.

gordon-quad, 2019-03-05:
Just for academic interest, what actually stops you from modding your iron so it can pump 100 W instead of 65? It looks to me that this zener clamping is completely artificial since RT7272 input voltage can be as high as 36V according to datasheet, the only reason for it I see is 25V rated caps. Does it mean if I replace SMPS input caps C8/C9, zener D3 for 30V clamping and optionally R20/R21 resistor divider to not saturate ADC I can use up to 30 V (breakdown threshold for MOSFET) DC input or am I missing something?

Ralim, 2019-03-06:
@gordon-quad 
You would need to replace the caps, the mosfet, modify the pcb with larger traces, reduce the contact resistance on the contacts to the tip, and possibly modify the inside of the tip as its rated for around 3A duty, and adjust the feedback resistors. The design is rated for 3A, you would be approaching 3.5A so you most likely would be fine, but I wouldnt be surprised if random things started to fail.
(30V would give around 3.4A on an 8.8 ohm total tip+mosfet resistance)

Honestly, if you *really* want that 100W, just use a 100W iron in the first place.

gordon-quad, 2019-03-06:
Nevermind, I read somewhere that T12 Hakko tips are 140 W rated but it appears that this information is not correct.

Firebie, 2021-01-05:
Sorry, which size of zener D3 is used - SOD-123F(case length 2.5mm, 500mW)  or SOD-323F (case length 1.6mm 200mW)?
From photo looks like SOD-123F.
Thanks