Ralim/IronOS #112: CAL VIN *jump* VIN Calibration

kokein, 2017-10-14:
Hi Ralim,

I found that there is lack of one "point" of number while VIN Cal setup.
I have the exactly 24V from my power adapter, but when entering VIN Cal. there is no 24V to select :)
It jumps from 29,9V up to 24.1V ...  Or am I doing something wrong?

Ralim, 2017-10-16:
Hi,
This is just how it currently works.
The ADC in the STM32 is sampling the incoming data in the range of 0-32768.
Then this is divided down by an integer factor to create a number that is stored in 100's of mV. (ie 19.5V = 195).

When you are adjusting the input you are adjusting the division factor that divides the data into the 100mV number. Because this is integer maths, there is no perfect solution for every single iron, and instead, it is just a case of selecting the *best* option for your iron :)

kokein, 2017-10-17:
Ok. Not a big deal ;) thx

closed 2017-10-17 by Ralim