Ralim/IronOS #113: Feature request: setting button sequance

gidimiz, 2017-10-14:
Hi.
I just tested v2.0-alpha.4.
When in the settings, its great that you can see what each option is if you wait x sec. but:
1. To exit you need to press one of the buttons. if you press the tip one, it exits and changes the option you had. if you press the back, it exits and moves to the next option. the best will be: if you press any key, it will exit the long description and will wait for the user to decide. 
2. In the long description, there should be also what option is doing what. in some of them, its not fully explained, like in the BOOST.
3. If you want to exit, you need to scroll all the way to the end. When you reach the end. have a small delay of 1-2 sec. For example, if i have finished changing the settings, I can then scroll fast to start the iron, and not worry that it will enter settings again.

Thanks, and keep the great work you are doing!

Ralim, 2017-10-16:
Hi,
Number 1 is certainly a bug that shouldn't be there.

 Number 2 is up to both users to suggest corrected descriptions or the translators for the language translations.

Number 3 already happens, if you keep holding the scroll key it won't re-enter the settings. It will, however, finish the settings and then enter the about screen that has the firmware version, but you can use any key to exit that screen easily.

Ralim, 2017-10-28:
Hi,
In the newer alphas (5+) Number 1 should be fixed :)

gidimiz, 2017-10-28:
Thanks for the quick update. I will check this and update you if I have any comments.

closed 2017-11-04 by Ralim