Ralim/IronOS #114: "flickering" values

noxhirsch, 2017-10-21:
Hi,
at first I have to say great firmware - thank you very much! 👍 
I noticed that the temperature is often switching between two values (e.g. 219°C and 220°C) several times a second (I don't know how fast it's changig, but it feels very fast 😉) and the infill of the battery icon is "jumping" up and down.
Maybe it's possible to smooth the values a little bit?

Ralim, 2017-10-22:
Hi @noxhirsch,

The screen updates at approximately 20Hz at the moment and the tip temperature is measured at 10Hz.
There is a filter that is adding around half a second of averaging to the temperature readings (Since they are extremely noisy). I can increase the display filter easily (its decoupled from the PID).

I personally prefer the rapid update rate of the screen as it shows the trend of the tip when you place it on a large pad. But I can certainly slow down the filter a little bit :)

noxhirsch, 2017-10-22:
Hi @Ralim,
slowing down the filter a little bit would be really great! 👍 
I don't know if it's possible (without a lot of work), but maybe you could average the battery display even more, so it is more stable (if it's decoupled from the temperature averaging)

Ralim, 2017-10-29:
@noxhirsch I did slow down the filters a little bit in the alpha 6 release, and would be curious to see your thoughts on this?

noxhirsch, 2017-10-29:
@Ralim The battery is ways better 👍 The temperature is still flickering often ;) But I had an idea just in this second: Maybe if the temperature is set to 320°C you could display 320°C even if it's 319/321. As soon as the temperature differs more than 1°C from the preset, the real value will be shown. So you get a stable display when the temperature is at the right point, but you still get fast updates when the temperature drops more then one degree. IMHO +/- 1°C is an acceptable tolerance.
What do you think?

aziemer, 2017-10-29:
@Ralim The averaging of the displayed temperature (Modes.c line 461ff) is "sub-optimal" ;-)
You should exchange lines 461 and 462 to really make this a 3 sample averaging - as it stands now it's only 2 sample averaging as you first assign the current temp to the middle sample and then this already changed value gets assigned to the last sample.

Ralim, 2017-10-30:
@aziemer You are looking at the 1.x C code, this is for the 2.x CPP code base. (1.x is no longer being bug fixed).

@noxhirsch Yeah thats a possible idea. Ill try it out and get back to you :)

closed 2018-01-09 by Ralim