Ralim/IronOS #115: DC LOW when connecting DC powesupply, fixes after calibration

LexioTech, 2017-10-28:
Hi,

With the recent build given to us to fix the "no sleep no auto off" bug for certain people, If I disconnect and reconnect the power to my TS100, and try to turn it into heating mode it says "DC LOW", after this error if I go to the "CAL VIN" menu, enter -> wait until the voltage rises to the already calibrated value (19.8 V), and then exit the menu with clicking the 2 buttons together, then I can start heating the iron without changing anything. Just visiting the CAL VIN menu somehow fixes this issue.

This did not occur with the previous version of the FW, and I'm using the same stock power supply that came with the TS100.

If you need any video for this, or debugging help, I'm more then happy to provide help :)

Thanks!

Ralim, 2017-10-28:
Hi,
This is known in that tester version I uploaded, the filtering on the DCinput isn't setup perfectly yet. I have a new alpha coming soon that fixes this :)

Ralim, 2017-10-28:
@LexioTech Should be fixed in Alpha 6

commit fd922711 (2017-10-28): Patching Filter Depth

Fixes #115
closed 2017-11-04 by Ralim