Ralim/IronOS #116: Bug report (I think) temperature not dropping all the way to sleep V2 Alpha 6

csdpm, 2017-10-29:
Hi Ben,

I've just noticed that the indicated tip temperature isn't dropping all the way to the sleep temperature. Sleep temperature is set at 150'C (default) but after the sleep timeout the temperature is dropping down and hovering between 167-169'C. I haven't seen this before. Standby timeout is set at 5 mins so there's plenty of time for it to drop.

My default screen is set to detailed, other screens to simple.

Ralim, 2017-10-30:
Hi,
Can you confirm what power source you are using? At higher input voltages its hard to regulate the temperature at the lower end? Ill try and replicate this locally as well. :)

csdpm, 2017-10-30:
Hi,

Yes sorry I should have included that. 24V (3A capable) on the nail. The iron was reading 23.6V until I calibrated it (I think it read correctly previously). If it's just mine then it's not a big issue, I can probably just up the sleep temp to 180 instead.

Ralim, 2017-10-30:
Ill verify locally and get back to you, having the higher voltage makes the PID more unstable at low input voltages (just because tiny duty cycle still raises the tip massively).

csdpm, 2017-11-03:
It looks as if your PID instability thought is in the right direction... I've tried changing the sleep temperature up to 170'C and the temperature settles at ~187'c. Dropping the sleep set point down to 140'C causes the temperature to settle at ~160'C.

matmemo, 2017-11-05:
I can confirm this too. I'm using a bench power supply at ~24V and the temperature drops to 167-169°C and stays there (sleep temp set to 150°C) like csdpm said above. Curiously it stays at the set 150°C if I force it to cool down that far by cooling the tip on a wet sponge.

For the time being I did just set the sleep temp to 130°C which makes it drop to ~150°C in air when the sleep mode is active.

matmemo, 2017-11-06:
Ok, I did some more testing and found that the problem really isn't that noticeable when using input voltages ~20V and a sleep temperature of 150°C. But at least for my device it's still there for lower temperatures and voltages. The easiest way to reproduce it is to set a low temperature in soldering mode and use the boost to drive the temperature up by 20-30°C.
When I let go of the button it starts to cool down but somewhere above the target temperature (I think it's about target+18°C) the heating icon starts flashing up for very short amounts of time and the temperature decreases very slowly from then on.

What seems to differ with the input voltage is the set temperature where it doesn't reach the target temperature when cooling down:
At 12V, 60°C set, flashing at 78°C,  settles at 68-69°C.
At 20V, 130°C set, flashing at 148°C, settles at 137-139°C.
At 24V, 160°C set, flashing at 178°C, settles at ~176°C.

These seem to be the upper bounds for this behaviour with my device.

When approaching the target from below the PID controller seems to work very good in any case.

I don't know very much about the algorithm. Would it make sense to simply disable the output driver when the tip is hotter than the target temperature, so it would cool down as fast as possible and set in again if heating is needed?

csdpm, 2018-01-01:
> Should help with #116

Indeed it does Ben. Mine seems to be holding temperature fine at Sleep temperature. I don't know if you want to wait for further confirmations before closing.

Many thanks.... and a Happy New Year to you.

closed 2018-04-27 by Ralim