Ralim/IronOS #117: Sleeping iron should not heat if sleep temperature > set tip temperature

tigert, 2017-10-30:
So I set the iron to 50C to test whether it would work as a hand warmer inside a transmitter glove (and it might work, 50C is a bit uncomfortable to touch with cold hands, but if there was a holder that keeps it away from fabric and fingers, why not) - it's a crazy use case but more geek points..

However there is a logical hazard in the current code: the moment the iron goes to sleep, it aims for the STMP (sleep temperature) set in the config. By default this is 150°C.

We can debate whether a hand warmer is a sensible use case, but in any case, the iron should maintain the set temperature also during sleep, if it is set lower than the STMP.

Of course we could do it so that the sleep temperature actually is determined dynamically from the set temperature, which might work better overall.

Ralim, 2017-10-30:
@tigert 
As you well know there is no enforcement of this at the moment, the user can have sleep and boost temperatures set to whatever they like. For this style of use case would it not make more sense to just set MSENSE to 0 to disable sleep and shutdown?

I can add the logic to enforce Boost>set>sleep if really desired?

tigert, 2017-10-31:
Good points. Yeah, after reading your #118 reply I agree this requires more thought. 

I think the logic to prevent sleep heating the iron hotter than you expect might be a good failsafe catch, though?

SijmenSchoon, 2017-11-01:
To me, the most logical solution would be setting the iron temperature to `min(systemSettings.SleepTemp, systemSettings.SolderingTemp)` when sleeping.

Ralim, 2017-11-02:
@SijmenSchoon That's exactly what I'm planning on adding to the firmware. Seems to be the safest and easiest option.

closed 2017-12-12 by Ralim