Ralim/IronOS #118: dynamic sleep temperature?

tigert, 2017-10-30:
This is actually an afterthought to #117 (a problematic case if sleep temperature is set higher than soldering temperature)

When there is a fixed sleep temperature, it will be optimal for roughly just one soldering temperature. Of course this is most commonly the case too - when soldering you might just use one temperature ever. And boost never triggers sleep anyway because you have the iron on your hand.

What if the sleep temperature was defined dynamically from the soldering temperature? So that it would be for example, say, 0.75 * soldering temperature, and then you just had the option to turn off automatic sleep functionality for the other weird use cases like heating your tea? ;-) 

When thinking about the sleep functionality, there are two points that I can think of:

1. Energy saving - so the temperature needs to be cooler than the soldering temperature, so it will save energy over resting periods compared to maintaining the soldering temperature all the time.
2. Speed of recovery - the sleep temperature should not be too low, so that the iron is hopefully ready to solder quickly when you pick it up.

Actually the sleep temperature drop should perhaps be somehow adjusted to the heating speed - the tip recovers much faster on lower temperatures, so the difference could be even higher (or sleep could even turn heating totally off for low soldering temperatures)

Ralim, 2017-10-30:
This is not left dynamic as you want to ensure that the sleeping temperature always remains below the temperature that the tip oxidises at, as well as below the melting temperature of solder (63/37 is 183C).

We don't want people to make the mistake of setting their soldering temp to ~400C and then have their sleep temp suddenly raised to ~190C which means the iron will oxidise and the solder will stay molten.

The sleep temperature is set to 150C as this avoids the solder staying molten, and avoids excessive oxidisation of the tip. This temperature was not chosen because of energy saving or speed of recovery. Usually, once the tip is up to temperature the power usage difference really isnt all that much until you start soldering.

Also keep in mind your point 2 is entirely operating voltage dependant.

tigert, 2017-10-31:
Good point! :-) I did not realize the oxidation part. So then it actually does make sense to keep the sleep temperature somewhere around 150 indeed.

I would lean towards disabling the sleep logic when the temperature is set below the sleep temperature that, because it makes zero sense to heat the iron warmer when you sleep, and the tip won't oxidize no matter what. And if you want sleep logic, then you can just set a lower sleep temperature.

closed 2017-12-12 by Ralim