Ralim/IronOS #123: Rounding tip temp

pilatomic, 2017-11-06:
Hi,

First, thank you for all the hard work on this firmware !

Just updated to rev 2.0 yesterday, and I noticed it does now round the measured temperature when it less than 3° from target temperature (main.cpp:78).

The primary reason I switched from the official firmware was exactly because it was rounding the temperature.

Could this be disabled ? I fail to see a usecase where a wrong tip temperature is preferred to the actual temperature.

Ralim, 2017-11-06:
Hi,
This was included as a request from #114.
I'm not sold on it either, though 3 degrees is a tiny amount of rounding and is less than some commercial units perform.

It's up to debate if it stays in future versions or not.

pilatomic, 2017-11-06:
Hi,

Yeah I understand how the temperature "jumping" is not nice.

Maybe displaying scrolling numbers would be easier, that way there are some possible display position between integer values.

I crafted a small GIF to explain what I mean : 
http://pila.fr/wordpress/wp-content/uploads/demoScrollingTS100-1.gif

A temperature of 339.5°C would appear with half of 339 shown and half of 340.

What do you think ?

csdpm, 2017-11-06:
As an alternative suggestion, you could maybe reinstate the heater power level bar. That would allow you to see how hard the heater is having to work to maintain the tip temperature (the underlying reason for wanting minimal rounding) while still having a steady diplay. It would be simpler to implement than scrolling temperature too.

Just a thought anyway. I agree, the rounding does seem a bit too 'round' at the moment.

pilatomic, 2017-11-06:
Actually, I think the left icon could be used as power indicator. Currently it only has 2 states : < 10% PWM and > 10% PWM. Maybe it could act as a power bar, with the height of the top part being related to the PWM value ?

Still, "hiding" the true tip temperature doesn't seem a good idea to me

Ralim, 2017-12-12:
@pilatomic More than happy to make the icon more interactive if someone feels like making yet more bitmaps for the states of the icon :)