Ralim/IronOS #125: I cannot see the full menu ..

Vidalcris, 2017-11-08:
Just installed V2.0_fr and i cannot go after ADVIDL.. so i cannot access boost settings and more.
I even tried to reinstall the original firmware and install again V2.0 but the problem is still there.

Ralim, 2017-11-08:
Is the problem only present in the _fr version?

Vidalcris, 2017-11-08:
Maybe yes.
I have to say i'm tired to have to do all the settings again on each flash ;)
This is why i just flashed alpha 6 XD

Vidalcris, 2017-11-08:
Ok i confirm that the problem is not present with V2.0 (not french)

ponchick, 2017-12-02:
Same issue with russian firmware. English version is Ok.

PenGenKiddy, 2017-12-02:
@ponchick Yesterday i installed ru firmware. All works fine. Can you attach video with bug?

ponchick, 2017-12-03:
@PenGenKiddy I downloaded and installed latest TS100_ru.hex (V2.0) from [releases page](https://github.com/Ralim/ts100/releases). ADVIDL is a last menu item, here is [video](https://www.dropbox.com/s/003y63r7uev76ai/2017-12-03%2011.51.19.mp4?dl=0).

sotima, 2017-12-05:
Same here with german Version of Release 2.0 (TS100_de.hex). Menu does  only go to ADVIDL.

commit 17e7d924 (2017-12-12): Fix Menu not handling some languages

Should close #125
closed 2017-12-12 by Ralim