Ralim/IronOS #130: Turkish Language Support

JustNoLimit, 2017-11-15:
#ifdef LANG_TR
const char* SettingsLongNames[16] = {
/*These are all the help text for all the settings.*/
/*No requirements on spacing or length*/
"Güç Kaynağı. kesim geriliminı ayarlar. <DC 10V> <S 3.3V hücre başına>",    //Power Source
		"Uyku Sıcaklığı <C>",    //Sleep Temp
		"Uyku Zaman Aşımı <Dakika/Saniye>",    //Sleep Timeout
		"Kapatma Zaman Aşımı <Dakika>",    //Shutdown Time
		"Hareket Hassasiyeti <0.Kapalı 1.En az duyarlı 9.En duyarlı>",    //Motion Sensitivity
		"Sıcaklık Ünitesi <C=Celsius F=Fahrenheit>",    //Temp Unit
		"Boş ekranda ayrıntılı bilgileri daha küçük bir yazı tipi ile göster.",    //Detailed Information
		"Görüntü Yönlendirme <A. Otomatik L. Solak R. Sağlak>",    //Orientation
		"Lehimleme yaparken ön tuşa basmak Boost moduna sokar(450C)",    //Boost enable
		"\"boost\" Modu Derecesi",    //Boost Temp
		"Güç verildiğinde otomatik olarak lehimleme modunda başlat. T=Lehimleme Modu, S= Uyku Modu,F=Kapalı",    //Auto start
		"Soğutma ekranında uç hala sıcakken derece yanıp sönsün.",    //Cooling Blink
		"Ucu kalibre et.",    //Calibrate Tip
		"Bütün ayarları sıfırla",    //Reset Settings
		"VIN Kalibrasyonu. Düğmeler ayarlar, çıkmak için uzun bas.",    //VIN Cal
		"Lehimleme yaparken detaylı bilgi göster",    //ADV SLD
		};

const char* SettingsCalibrationWarning = "Lütfen devam etmeden önce ucun oda sıcaklığında olduğunu garantiye alın!";
const char* UVLOWarningString = "LOW VOLT";    //Fixed width 8 chars
const char* SleepingSimpleString = "Zzzz";    // Must be <= 4 chars
const char* SleepingAdvancedString = "Uyuyor...";    // <=17 chars
const char* WarningSimpleString = "HOT!";    //Must be <= 4 chars
const char* WarningAdvancedString = "UYARI! UÇ SICAK!";

const char SettingTrueChar = 'T';
const char SettingFalseChar = 'F';
const char SettingRightChar = 'R';
const char SettingLeftChar = 'L';
const char SettingAutoChar = 'A';
#endif

wortex1979, 2017-12-10:
how can i update to turkish?

Ralim, 2017-12-11:
When I have time ill release a build file on the releases page with Turkish as an option :)

wortex1979, 2017-12-11:
thanks ve are waiting.

wortex1979, 2017-12-11:
#ifdef LANG_TR
const char* SettingsLongNames[16] = {
/These are all the help text for all the settings./
/No requirements on spacing or length/
"Güç Kaynağı. kesim geriliminı ayarlar. <DC 10V> <S 3.3V hücre başına>", //Power Source
"Uyku Sıcaklığı ", //Sleep Temp
"Uyku Zaman Aşımı <Dakika/Saniye>", //Sleep Timeout
"Kapanma Zaman Aşımı ", //Shutdown Time
"Hareket Hassasiyeti <0.Kapalı 1.En az duyarlı 9.En duyarlı>", //Motion Sensitivity
"Sıcaklık birimi <C=Celsius F=Fahrenheit>", //Temp Unit
"Boş ekranda ayrıntılı bilgileri daha küçük bir yazı tipi ile göster.", //Detailed Information
"Görüntü Yönlendirme <A. Otomatik L. Solak R. Sağlak>", //Orientation
"Lehimleme yaparken ön tuşa basmak Boost moduna sokar(450C)", //Boost enable
""boost" Modu Derecesi", //Boost Temp
"Güç verildiğinde otomatik olarak lehimleme modunda başlat. T=Lehimleme Modu, S= Uyku Modu,F=Kapalı", //Auto start
"Soğutma ekranında uç hala sıcakken derece yanıp sönsün.", //Cooling Blink
"Ucu kalibre et.", //Calibrate Tip
"Bütün ayarları sıfırla", //Reset Settings
"VIN Kalibrasyonu. Düğmeler ayarlar, çıkmak için uzun bas.", //VIN Cal
"Lehimleme yaparken detaylı bilgi göster", //ADV SLD

closed 2017-12-12 by Ralim

wortex1979, 2017-12-12:
tanks alot ralim. very good.

commit 121feef9 (2017-12-12): Add Turkish translation

Should close #130
Need to test glyph support