Ralim/IronOS #132: Polish support language

WerVa, 2017-11-20:
#ifdef LANG_PL
const char* SettingsLongNames[16] = {
/*These are all the help text for all the settings.*/
/*No requirements on spacing or length*/
"Źródło zasilania. Ustaw napięcie odcięcia. <DC 10V> <S 3.3V dla ogniw Li>",    //Power Source
		"Temperatura uśpienia <°C>",    //Sleep Temp
		"Czas uśpienia <Minuty/Sekundy>",    //Sleep Timeout
		"Czas wyłączenia <Minuty>",    //Shutdown Time
		"Czułość ruchu <0.Wyłączona 1.minimalna 9.maksymalna>",    //Motion Sensitivity
		"Jednostka temperatury <C=Celsius F=Fahrenheit>",    //Temp Unit
		"Wyświetla szczegółowe informacje za pomocą mniejszej czcionki na ekranie bezczynnośći <T = wł., N = wył.>",    //Detailed Information
		"Orientacja wyświetlacza <A. Automatyczna L. Leworęczna P. Praworęczna>",    //Orientation
		"Użyj przycisku przedniego w celu zwiększenia temperatury <T = wł., N = wył.>",    //Boost enable
		"Temperatura w trybie \"boost\" ",    //Boost Temp
		"Automatyczne uruchamianie trybu lutowania po włączeniu zasilania. T=Lutowanie, S= Tryb Uspienia ,N=Wyłącz",    //Auto start
		"Temperatura na ekranie miga, gdy grot jest jeszcze gorący. <T = wł., N = wył.>",    //Cooling Blink
		"Kalibracja temperatury grota lutownicy",    //Calibrate Tip
		"Zresetuj wszystkie ustawienia",    //Reset Settings
		"Kalibracja napięcia wejściowego. Krótkie naciśnięcie, aby ustawić, długie naciśnięcie, aby wyjść.",    //VIN Cal
		"Wyświetl szczegółowe informacje podczas lutowania <T = wł., N = wył.>",    //ADV SLD
		};

const char* SettingsCalibrationWarning = "Przed kontynuowaniem upewnij się, że końcówka osiągnela temperature pokojowa!";
const char* UVLOWarningString = "LOW VOLT";    //Fixed width 8 chars
const char* SleepingSimpleString = "Zzz!";    // Must be <= 4 chars
const char* SleepingAdvancedString = "Uspienie...";    // <=17 chars
const char* WarningSimpleString = "HOT!";    //Must be <= 4 chars
const char* WarningAdvancedString = "UWAGA! GORĄCA KOŃCÓWKA!";

const char SettingTrueChar = 'T';
const char SettingFalseChar = 'N';
const char SettingRightChar = 'P';
const char SettingLeftChar = 'L';
const char SettingAutoChar = 'A';
#endif

Ralim, 2017-11-20:
I will need to check all the glyphs used are present in the firmware. If they are I can add support for this shortly. :)

jonnieZG, 2017-11-22:
Polish alphabet falls under Windows-1250 encoding, the same as Croatian, Serbian, Slovenian and quite a few other Slavic languages. They are included in U+0100 to U+017F, or UTF-8 ranges C480-C4BF, C580-C5BF I've taken a look at the Font.h and OLED.cpp classes, and currently they are not supported.

The Russian Cyrillic font with letters defined in that range could be replaced during compile-time, depending on a selected language. I could make the font these days when I get some free time and propose a modification to the Font.h and OLED.cpp.

Ralim, 2017-11-23:
I'm happy to be replacing the characters at compile time that is no issue at all.
Just I'm over creating font maps for this project 😛 
Ill include the ASCII version from #135 first and then update to full font if anyone ever is bored enough to make more font glyphs :)

jonnieZG, 2017-11-23:
I'll do it.

Ralim, 2017-11-24:
This may be of help to you [Online Editor](http://ipbox.linkpc.net/tsfed/).
Massive thanks to @uncle-fed from #87

jonnieZG, 2017-11-24:
Great! I was using this one: https://www.pentacom.jp/pentacom/bitfontmaker2/ with a Java utility that I made for converting between the two formats, but this one by @uncle-fed I like even better.

jonnieZG, 2017-11-24:
I'm done. With the following update, all the Slavic languages will be able to be fully supported, including Czech, Slovak, Polish, Croatian, Serbian, Bosnian, Slovenian and others that use Latin script.

It supports characters from U+0100 to U+017F.

If LANG_RU is enabled, then it will define Cyrillic script. In all other cases, the alternative characters from the U+0100 to U+017F range will be defined. Please not that in the latter case, the font space uses 384 bytes more than the version with the Cyrillic script, due to a bigger character range.

**Translation.h**
```C
/*
 * Font.h
 *
 *  Created on: 17 Sep 2016
 *      Author: Ralim
 *
 *      ... This file contains the font...
 */

#ifndef FONT_H_
#define FONT_H_
#include "Translation.h"
/*
 * Remember screen is LSB at the top, MSB at the bottom of the strip!
 */
const uint8_t FONT_12[]={
		0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,//blank
		0x00,0x00,0x00,0x00,0x7C,0xFF,0xFF,0x7C,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x33,0x33,0x00,0x00,0x00,0x00,0x00,//!
		0x00,0x00,0x00,0x3C,0x3C,0x00,0x00,0x3C,0x3C,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,//"
		0x00,0x00,0x10,0x90,0xF0,0x7E,0x1E,0x90,0xF0,0x7E,0x1E,0x10,0x00,0x02,0x1E,0x1F,0x03,0x02,0x1E,0x1F,0x03,0x02,0x00,0x00,//#
		0x00,0x00,0x78,0xFC,0xCC,0xFF,0xFF,0xCC,0xCC,0x88,0x00,0x00,0x00,0x00,0x04,0x0C,0x0C,0x3F,0x3F,0x0C,0x0F,0x07,0x00,0x00,//$
		0x00,0x00,0x38,0x38,0x38,0x00,0x80,0xC0,0xE0,0x70,0x38,0x1C,0x00,0x30,0x38,0x1C,0x0E,0x07,0x03,0x01,0x38,0x38,0x38,0x00,//%
		0x00,0x00,0x00,0xB8,0xFC,0xC6,0xE2,0x3E,0x1C,0x00,0x00,0x00,0x00,0x00,0x1F,0x3F,0x31,0x21,0x37,0x1E,0x1C,0x36,0x22,0x00,//&
		0x00,0x00,0x00,0x00,0x27,0x3F,0x1F,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,//'
		0x00,0x00,0x00,0xF0,0xFC,0xFE,0x07,0x01,0x01,0x00,0x00,0x00,0x00,0x00,0x00,0x03,0x0F,0x1F,0x38,0x20,0x20,0x00,0x00,0x00,//(
		0x00,0x00,0x00,0x01,0x01,0x07,0xFE,0xFC,0xF0,0x00,0x00,0x00,0x00,0x00,0x00,0x20,0x20,0x38,0x1F,0x0F,0x03,0x00,0x00,0x00,//)
		0x00,0x00,0x98,0xB8,0xE0,0xF8,0xF8,0xE0,0xB8,0x98,0x00,0x00,0x00,0x00,0x0C,0x0E,0x03,0x0F,0x0F,0x03,0x0E,0x0C,0x00,0x00,// *
		0x00,0x00,0x80,0x80,0x80,0xF0,0xF0,0x80,0x80,0x80,0x00,0x00,0x00,0x00,0x01,0x01,0x01,0x0F,0x0F,0x01,0x01,0x01,0x00,0x00,//+
		0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0xB8,0xF8,0x78,0x00,0x00,0x00,0x00,0x00,//,
		0x00,0x00,0x80,0x80,0x80,0x80,0x80,0x80,0x80,0x80,0x00,0x00,0x00,0x00,0x01,0x01,0x01,0x01,0x01,0x01,0x01,0x01,0x00,0x00,//-
		0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x38,0x38,0x38,0x00,0x00,0x00,0x00,0x00,//.
		0x00,0x00,0x00,0x00,0x00,0x80,0xC0,0xE0,0x70,0x38,0x1C,0x0E,0x00,0x18,0x1C,0x0E,0x07,0x03,0x01,0x00,0x00,0x00,0x00,0x00,// /

		0x00,0xF8,0xFE,0x06,0x03,0x83,0xC3,0x63,0x33,0x1E,0xFE,0xF8,0x00,0x07,0x1F,0x1E,0x33,0x31,0x30,0x30,0x30,0x18,0x1F,0x07,//0
		0x00,0x00,0x00,0x0C,0x0C,0x0E,0xFF,0xFF,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x30,0x30,0x30,0x3F,0x3F,0x30,0x30,0x30,0x00,//1
		0x00,0x1C,0x1E,0x07,0x03,0x03,0x83,0xC3,0xE3,0x77,0x3E,0x1C,0x00,0x30,0x38,0x3C,0x3E,0x37,0x33,0x31,0x30,0x30,0x30,0x30,//2
		0x00,0x0C,0x0E,0x07,0xC3,0xC3,0xC3,0xC3,0xC3,0xE7,0x7E,0x3C,0x00,0x0C,0x1C,0x38,0x30,0x30,0x30,0x30,0x30,0x39,0x1F,0x0E,//3
		0x00,0xC0,0xE0,0x70,0x38,0x1C,0x0E,0x07,0xFF,0xFF,0x00,0x00,0x00,0x03,0x03,0x03,0x03,0x03,0x03,0x03,0x3F,0x3F,0x03,0x03,//4
		0x00,0x3F,0x7F,0x63,0x63,0x63,0x63,0x63,0x63,0xE3,0xC3,0x83,0x00,0x0C,0x1C,0x38,0x30,0x30,0x30,0x30,0x30,0x38,0x1F,0x0F,//5
		0x00,0xC0,0xF0,0xF8,0xDC,0xCE,0xC7,0xC3,0xC3,0xC3,0x80,0x00,0x00,0x0F,0x1F,0x39,0x30,0x30,0x30,0x30,0x30,0x39,0x1F,0x0F,//6
		0x00,0x03,0x03,0x03,0x03,0x03,0x03,0xC3,0xF3,0x3F,0x0F,0x03,0x00,0x00,0x00,0x00,0x30,0x3C,0x0F,0x03,0x00,0x00,0x00,0x00,//7
		0x00,0x00,0xBC,0xFE,0xE7,0xC3,0xC3,0xC3,0xE7,0xFE,0xBC,0x00,0x00,0x0F,0x1F,0x39,0x30,0x30,0x30,0x30,0x30,0x39,0x1F,0x0F,//8
		0x00,0x3C,0x7E,0xE7,0xC3,0xC3,0xC3,0xC3,0xC3,0xE7,0xFE,0xFC,0x00,0x00,0x00,0x30,0x30,0x30,0x38,0x1C,0x0E,0x07,0x03,0x00,//9
		0x00,0x00,0x00,0x00,0x70,0x70,0x70,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x1C,0x1C,0x1C,0x00,0x00,0x00,0x00,0x00,//:
		0x00,0x00,0x00,0x00,0x70,0x70,0x70,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x9C,0xFC,0x7C,0x00,0x00,0x00,0x00,0x00,//;
		0x00,0x00,0xC0,0xE0,0xF0,0x38,0x1C,0x0E,0x07,0x03,0x00,0x00,0x00,0x00,0x00,0x01,0x03,0x07,0x0E,0x1C,0x38,0x30,0x00,0x00,//<
		0x00,0x00,0x60,0x60,0x60,0x60,0x60,0x60,0x60,0x60,0x60,0x00,0x00,0x00,0x06,0x06,0x06,0x06,0x06,0x06,0x06,0x06,0x06,0x00,//=
		0x00,0x00,0x03,0x07,0x0E,0x1C,0x38,0xF0,0xE0,0xC0,0x00,0x00,0x00,0x00,0x30,0x38,0x1C,0x0E,0x07,0x03,0x01,0x00,0x00,0x00,//>
		0x00,0x1C,0x1E,0x07,0x03,0x83,0xC3,0xE3,0x77,0x3E,0x1C,0x00,0x00,0x00,0x00,0x00,0x00,0x37,0x37,0x00,0x00,0x00,0x00,0x00,//?
//V 32
		0x00,0xF8,0xFE,0x07,0xF3,0xFB,0x1B,0xFB,0xFB,0x07,0xFE,0xF8,0x00,0x0F,0x1F,0x18,0x33,0x37,0x36,0x37,0x37,0x36,0x03,0x01,//@
		0x00,0x00,0x00,0xE0,0xFC,0x1F,0x1F,0xFC,0xE0,0x00,0x00,0x00,0x00,0x38,0x3F,0x07,0x06,0x06,0x06,0x06,0x07,0x3F,0x38,0x00,//A
		0x00,0xFF,0xFF,0xC3,0xC3,0xC3,0xC3,0xE7,0xFE,0xBC,0x00,0x00,0x00,0x3F,0x3F,0x30,0x30,0x30,0x30,0x30,0x39,0x1F,0x0F,0x00,//B
		0x00,0xF0,0xFC,0x0E,0x07,0x03,0x03,0x03,0x07,0x0E,0x0C,0x00,0x00,0x03,0x0F,0x1C,0x38,0x30,0x30,0x30,0x38,0x1C,0x0C,0x00,//C
[... cut ...]
void OLED::drawArea(int16_t x, int8_t y, uint8_t wide, uint8_t height, const uint8_t* ptr) {
	// Splat this from x->x+wide in two strides
	if (x < 0)
		return;    //cutoffleft
	if ((x + wide) > 96)
		return;    //cutoff right
	if (y == 0) {
		//Splat first line of data
		for (uint8_t xx = 0; xx < (wide); xx++) {
			firstStripPtr[xx + x] = ptr[xx];
		}
	}
	if (y == 8 || height == 16) {
		// Splat the second line
		for (uint8_t xx = 0; xx < wide; xx++) {
			secondStripPtr[x + xx] = ptr[xx + (height == 16 ? wide : 0)];

		}
	}
}
```

Mrkvozrout, 2017-12-02:
@jonnieZG I am just testing the new font, it works for most part but not completely.

All but the first line works good. In the first line
`U0100 0Ā 1ā 2Ă 3ă 4Ą 5ą 6Ć 7ć 8Ĉ 9ĉ AĊ Bċ CČ Dč EĎ Fď`

- letters 0-A shows just completely fills the whole letter space
- letter B shows just the upper half of the letter space filled

[image]

**Note 1**
Updated font does not contain all characters needed to fully support Czech (and probably other languages - thinking german with umlauts). Also `00C0-00FF` would be needed. Hopefully also the `00B0` character to allow °C and °F be used in translation texts.

**Note 2**
In your posted modification from @jonnieZG , there is a corrupt `o` definition. Look for `//V64` section. There is a `alc` instead of char definition - I assume this is an unintended mistake.

@Ralim I propose accepting current waiting languages (Polish, Czech..) without diacritics and creating new issue for this extended font. Then languages can be updated to use diacritics.

jonnieZG, 2017-12-05:
@Mrkvozrout You are right about the letter "o" definition, it was obviously an unintentional error. I corrected it now in the section above, and I expect you did the same before compiling and testing it. ;)

Regarding U+00C0 to U+00FF they *are* already supported, and I did not mess with that part. I might be wrong, but if you tested them and did not get them to work, I'd say that your source code was not in UTF-8. Please check in Eclipse (or whatever dev software you use) if it is set correctly. That might explain why the other characters don't work either. Even though we reference here characters by Unicode, the program expects them to be encoded in UTF-8.

Unfortunately, I don't have a development environment for this device installed, so I can not test it myself.

jonnieZG, 2017-12-06:
WoskStation for STM32 ;)

uncle-fed, 2017-12-06:
Have a look here also: https://github.com/Ralim/ts100/issues/87#issuecomment-349776478

Mrkvozrout, 2017-12-07:
So I did some more testing. Obviously I have not tested all the letters but most of those I did test works fine. But not all - for me.

The reason I write "for me" is e.g. this:
[image]

As you can see, the letter `à` is corrupted. I imported its definition to the [font editor](http://ipbox.linkpc.net/tsfed/) but there it looks ok. I switched the letter with regular `a` and bam... it looked ok even on the TS100 and now the regular `a` was corrupted in the same way.

I observe similar issue with `Ā` letter. It just displays as full white square (see my previous photo) and again in editor it was ok. When I switched it for `b`, it displayed ok and `b` showed as filled square.

**I need someone else to try this on their device**. Maybe I have corrupted/degraded internal memory (after all this firmware flashing)? Or could it be just something with the array definition (looks ok to me, no extra bytes that I noticed).

**Of course for the `C2` section to work, it is needed to do the bugfix** pointed out by @jonnieZG (also found it :) ). This is because of the skipped 32 control characters so the `C2` section needs to compensate for this. Other sections are ok.

Ralim, 2017-12-12:
I have merged in this.
Thank you for all your work :)
Looks to be compiling and running ok for me here, though as I cant actually read the language, I could totally be missing small errors.

Re-open the issue or hit me up if there is anything I can do to help :)

commit 39295c97 (2017-12-12): Adding Polish support

Closes #132 afaik.
closed 2017-12-12 by Ralim