Ralim/IronOS #134: Proposal: "+" and "-" instead of "T" and "F"

jonnieZG, 2017-11-22:
In menus, it might be more intuitive to use "+" and "-" instead of "T" and "F".

Mrkvozrout, 2017-11-22:
Thinkin about your proposal with conclusion that I would be more confused to see +- than TF as the latter is commonly known as true/false (at least in my circles).

But what about to use the proper accepted/rejected signs like **✓** (unicode "check mark") and **✗** ("ballot X") ?

jonnieZG, 2017-11-22:
I am quite a nerd, but in this context True and False are not the first things that come to my mind. Especially when you have translations to different languages. As you suggest, ballot signs would also be a better choice. I personally prefer + and -, but that's just me. ;) Whatever the majority settles about.

Ralim, 2017-11-23:
I am happy to swap to tick and cross marks. +- is fairly illogical to most people I have asked about their thoughts about this locally. Can I ask @jonnieZG what your cultural background is ?

jonnieZG, 2017-11-23:
I am an electrical engineer and Java architect, my mother tongue is Croatian, and am fluent in Italian and English. I read and write a lot. Besides working on many professional and hobby projects (Star Wars fandome related), I am also an official translator for [Waze](https://www.waze.com/), [Live for Speed](https://www.lfs.net/) and a few other, smaller products.

Ralim, 2017-11-23:
Woah, sorry I didn't mean to insult you I was just curious as to why you would prefer +/-. In translations the T/F can be changed to other symbols if preferred. Just to some people +/- does not correlate with an on/off notation if that makes sense?

jonnieZG, 2017-11-23:
Don't worry, no offense taken. It is normal to want to know some background of the people who are messing into your business! ;)

Well for me,, + and - symbols are as much intuitive as it gets, if we constrain on just one symbol per state, and exclude other special symbols. Comparing + and - it is clear that a "+" symbolizes something "with" and a "-" something "without". 

On the other hand, 0 and 1 might suggest a count that is not necessarily constrained to the binary digits. while using ballot and check signs requires defining them in a font, and complicating the manuals, because you need to reference them using the same symbols. "+" and "-" are always available, and are non-ambiguous to pronounce.

Ralim, 2017-11-25:
Yeah, I just find +- to be really confusing so I was curious what your background was :)

Honestly, this might become a language-specific change, since to some cultures T/F is more logical than +- ?

Mrkvozrout, 2017-11-25:
Isn't it already language specific?
I mean in the Translation.c there are:
```
SettingTrueChar = 'T';
SettingFalseChar = 'F';
SettingRightChar = 'R';
SettingLeftChar = 'L';
SettingAutoChar = 'A';
```

Ralim, 2017-11-25:
Yes it is, but I mean that it might be that some languages/cultures will have this set to +/- and some T/F etc ?

Mrkvozrout, 2017-11-25:
I agree. I already used this possibility in the czech translation.

jonnieZG, 2017-11-25:
+1 for that

Ralim, 2017-12-12:
Closing this as its a language-specific change that the whole world will *not* agree on.

Feel free to submit pull requests for languages where this is more logical.

closed 2017-12-12 by Ralim