Ralim/IronOS #139: cannot access settings menu

canc3rmoon, 2017-11-25:
Hi,
Just got a brand new unit today with DFU is 3.43 and stock Firmware 2.12.
I've flashed the latest firmware (window10, thinkpad t530), apparently with success. Each time I get a .RDY result. Yet I cannot enter the settings when pressing the button closest to the tail. Flashing older firmwares resulst in showing Firwmare 2.12. Any suggestions pointing me in the right direction is highly appreciated.

jonnieZG, 2017-11-25:
Does it work with the old firmware? You first have to verify that the button itself is ok.

Mrkvozrout, 2017-11-25:
You can try the newest official firmware 2.17. Open following link and download the `APPV217.rar`
[www.minidso.com/forum](http://www.minidso.com/forum.php?mod=viewthread&tid=892&extra=page%3D1%26filter%3Dtypeid%26typeid%3D63)

canc3rmoon, 2017-11-25:
Thank you. The button does click, but does not open the settings menu on the orignially installed firmware. Flashing 2.17 results in .RDY, but upon booting shows fw 2.12 and still no access to menu.

jonnieZG, 2017-11-25:
So you have a faulty button, or a faulty connection. Either try to fix it yourself, or flash back the original firmware and activate your warranty. I'd go for the second option.

canc3rmoon, 2017-11-25:
Seems like I have a DOA. Will opt for warranty. Thx for your help.

closed 2017-12-12 by Ralim