Ralim/IronOS #142: Autostart to sleep mode

Mrkvozrout, 2017-12-03:
I noticed that in the description of autostart there is described option S, meaning start in sleep mode.

> Automatically starts the iron into soldering on power up. T=Soldering, S= Sleep mode,F=Off

But only true/false can be set in settings and code on startup also does not count with the S option.

Is this a leftover from 1.x version or should it be implemented?

Ralim, 2017-12-04:
Both?
In 1.x there were 3 modes, in 2.x there is only plan for the two modes as of current.
The third mode can be added if there is demand for it I guess? Its not hard to add.

PeterFalken, 2018-01-10:
I think that it would be a good addition - it can make it safer if accidentally plugged in.

Mrkvozrout, 2018-01-11:
@PeterFalken You mean safer in case you have explicitly selected to autostart to sleep temperature? I ask because autostart is disabled for default.

PeterFalken, 2018-01-11:
Yes @Mrkvozrout.
Setting the autostart to sleep mode and having the sleep temperatura be lower than the active temperature would be an ideal safety measure. It allows the iron to warm up to the sleep temperature.

SmayK, 2018-03-16:
hello yes my French translation is also based on this type of choice which is very good
for proof of my last correction
`/* Automatic start mode               */ "Choix de mode de démarage automatique à l'allumage <A=Activé, V=Mode Veille, D=Désactivé>",`

closed 2020-04-04 by Ralim