Ralim/IronOS #144: Fix reset settings menu item

Mrkvozrout, 2017-12-06:
`RESET` menu item does what it is supposed to do, but the implementation and user interaction does not make much sense.

Currently when clicking this menu item, there is `T`/`F` switching on screen. To reset settings, you must cycle to the end of menu where a flag is checked. This is the part that does not make much sense, even more because there is another settings item after the reset item. If you change the settings after you flagged the reset, it is effectively waste of time since the values are still getting to be reset.

Selection of `RESET` menu item should result in asking user for confirmation and proceed if confirmed, otherwise cancel and return to settings to last item (reset item). Also this should be the last menu item.

Mrkvozrout, 2017-12-06:
I have this already implemented and tested.

The behavior is unified with the `TMP CAL` option.

- user selects menu item
- warning/question is shown waiting for confirmation
- confirmed -> proceed with action (reset, calibrate..)
- canceled or timed out -> return menu item

I have not issued Pull request because I am not sure how to do the branches properly so that merging is not hell after the othe pull requests are (hopefully) accepted (still branching master? or wait for pull requests I count with?). So let me know how to approach this.

Ralim, 2017-12-12:
@Mrkvozrout 
I'm currently sitting here merging in the current pull requests right now, so hopefully should be easy to make a pull request shortly :)

I'm happy for me to go through pull request hell occasionally since my available time to work on this changes wildly, so it's my fault 😛

Mrkvozrout, 2017-12-16:
Implemented.

closed 2017-12-16 by Mrkvozrout