Ralim/IronOS #145: Wrong tip temperature reading

jorgempy, 2017-12-07:
Hello, first of all thank you ralim for this awesome firmware!

I'm having an issue with it though, i think the firmware is reading the wrong temp, because when I plug it in, the hot tip warning comes on, and it says that the tip is at ~145C, even when I have JUST plugged it in.

I've tried calibrating the temp, I've tried the v1.17 firmware, has a similar issue, but the cal at least fails, in the V2 firmware i can't know the result of the calibration, just the dots appear and thats it.

With the factory firmware the temp was fine, it started at 25C and it worked. I now have installed the minware V2.17 firmware and it also starts from 25C or so. Also when it heats up, the tip "feels" hotter, and it kind of makes sense if the ralim firmware is reading with a ~120C offset.

If anyone can point me to something else to try, I'm open to sugestions.
Thanks already.

Cpt-Huck, 2017-12-07:
Having no problems actually.. Using 2 different tips for the TS100.. Works fine..

jonnieZG, 2017-12-08:
I would say your tip is faulty, that is, the thermistor in the tip is faulty.

Regarding the calibration I opened an issue recently about the tip temperature displaying 0°C at room temperature, but I closed it after @Ralim explained that the temperature displayed is a difference between the tip and the handle temperature. 

However, I would still prefer **if the tip temperature after the calibration would be set to room temperature at a nominal 23°C**. That is a sane guess, but there could also be an option for manually tweaking that nominal room temperature. That would be a much better behavior, IMHO.

Ralim, 2017-12-10:
Hi,
The tip calibration must be run with the handle and tip at the same temperature. 

@jonnieZG Im not sure why yours is showing 0 after calibration as it shouldn't unless your tip is at 0C.

The way the tip calibration works is based on the nearly linear voltage response of the thermal couple Junction in the tip.

The code is programmed with the temperature response curve of the typical tip. However, since all tips are different they will have their own offsets.

As thermalcouples only actually measure relative to the cold Junction, which occurs in the iron, there is a second temperature sensor that is mounted near to this contact. 

When you perform calibration the unit takes the calculated temperature of the handle and compares this with the readings from the tip. Between these two the offset for the tip can then be calibrated. 

So again, I re- iterate, you absolutely must have the tip and handle at the same temperature. This is why there is the warning in the firmware. So if you have your handle warmer from being held or you have heated up the tip then you will have extremely large errors in your readings.

Again, the reading shown is the *actual* temperature.
If this is reading wrong please ensure you have performed the calibration correctly, with both your tip and the handle being the same. 

For almost all users the default calibration values should get you within a few degrees C. 

The new firmware will either show an error or give you the dots. If you get to the dots then your tip was cool enough to be possible. There isn't any nice way for the firmware to have a closed loop for calibration which is why there is the warning to ensure they are both at the same temperature.

jonnieZG, 2017-12-11:
@Ralim, thanks for the explanation, I get it. The soldering iron with the stock firmware shows the exact temperature after the calibration. This one, on the other hand, shows 0. I'll post a video.

Ralim, 2017-12-11:
@jonnieZG Please provide a video showing this fault.
The firmware should *not* start from 0.

jonnieZG, 2017-12-12:
Ok, I've filmed a short video, but there's no need for publishing it, since I've figured out the problem: the temperature sensor in the handle is loose, because as I slightly bend the handle, the temperature reading jumps to 27°C and then back to 0°C. I'll have to take it apart and solder the thermistor properly. 

@Ralim, do you by any chance have a schematic with the PCB layout, so I could locate the thermistor?

jorgempy, 2018-01-05:
So, i ordered a new tip, connected it and it had the same issue.

As @jonnieZG i took apart the device and found a pin of the TMP36GRTZ floating (maybe some shock during shipment). Resoldered and the Ts100 works!!.

Maybe an entry can be added to the wiki warning about this issue.

Thanks again @Ralim for your awesome firmware and @jonnieZG for figuring out this issue!

closed 2018-01-05 by jorgempy

Chilus, 2018-07-15:
Hi @jorgempy,
I'm currently experiencing a very similar problem as you described. When I plug in the cable the tip temperature immediately jumps to ~240C. I also don't see a boot logo or any of that sorts. I have original tips en tried with several of them, same problem occurs. 

The strange thing is that opened up the solder iron to check for loose pins on the TMP36GRTZ sensor, but there wasnt any. I resoldered the 5 pins and measured everything through but the problem remains.

Do you have any thoughts on what I might be missing here?

Bulltac, 2019-02-10:
I am having the same problem as the OP and "Chilus." I feel like this issue is relatively common and the thread was high jacked by the guy with the "0C" problem

whitehoose, 2019-02-10:
I don't see even gross variances as a "problem" as long as it's predictable.  Anyone doing temp critical work is going to have (or should have) the kit to accurately measure tip temp direct - offset measurements are notoriously prone to misleading returns.  I "see" the temp display and as long as it returns expected behaviour I don't consider that actual values.  The temp could be a dotted line for me ..............  as long as I have some indication.  If it began swinging unpredictably or hung below "usual" levels - I'd know something was amiss ... but if the melt was consistent I'd ignore the indicator.  

It's handy to have an inkling that the iron is getting hot at a reasonable rate and that it's reached a temp that isn't dangerous to life an limb but it's more dangerous to regard it as anything other than an indicator, It's certainly not a measurement.  No one I know (who isn't supported by pukka instrumentation) sticks to a tip temp of x.  You melt solder and look for all the signs that a wet finger measurement gives you.  If wire looks a bit thick - you crank up the heat a little, and try again.  I'd as soon plug in a usb tester.  jonnieZG's problem seems potentially far more serious.

Bulltac, 2019-02-10:
If its off by over 100c, and if the iron is between 50 and 100c LOWER than its set temp than i am missing out on a huge portion of my irons potential heat. Even with the extra 50c this firmware gives you. I had to flash mine back to minware because i would have to have the iron set to 350c just to melt 180c solder, and max out at 450c just to do the larger joints that my cheapo desk iron does at 300c. You arw right that it doesnt matter, IF the iron stil has access to its full tempurature range. If not, then it has crippled the iron.

whitehoose, 2019-02-10:
That's where the intelligence(?) of "smart" devices crumbles.  My old Antex or Weller beat this new crop  hands down - they never questioned my decisions.  I started with a stove and copper slug, progressed to gas and electrickery, I got my first iron with temperature control 3 years ago and honestly the the quality of work hasn't benefited.  Quite the contrary, unless I can provide all the complex and expensive kit required to "prove" my power is worthy of entering my Iron's innards I can't even get the bugger to switch on, and if I can, that's no guarantee that it will melt solder.
Progress indeed!  If the zombie apocalypse starts during a powercut - we are screwed.

Ralim, 2019-02-10:
@Bulltac 
Please provide all of the measurements from the debugging menu in the latest firmware please, to see if your ADC trim is off or if it's rated to the tmp ic.

Bulltac, 2019-02-10:
I submitted a bug report but i dont think i included that info. I will get that data for you as soon as i get home. Thank you so much for your quick response. I bought the ts100 specifically to run your firmware so it would be awesome to get it working. Thank you for all your work on this man it makes the iron unparralelled.

Chilus, 2019-02-11:
@Bulltac I still suspect it is some kind of a hardware problem because I bought another TS100 and Ralim's firmware works perfectly fine on that one. @Ralim How would you know if the ADC trim is off? Can you explain how to test this?

Ralim, 2019-02-11:
@Chilus #451 Ill comment more over there