Ralim/IronOS #148: Convert CasioNostalgia Font

Ralim, 2017-12-12:
To enable support for languages other than english to use the small fonts, we need a new font.
Over in Issue #87 @uncle-fed has recieved confirmation from the creator of [CasioNostalgia ](https://fontstruct.com/fontstructions/show/1193786/casionostalgia) is okay with it being used with GPLv3.

Someone needs to convert this font map over to the format that is used by the OLED screen in the TS100.

Not sure when I'll have time, so leaving this open to anyone who wants the challenge.

perillamint, 2017-12-12:
Wait, isn't it BY-**NC**-SA?

Ralim, 2017-12-12:
@uncle-fed said in the [comment](https://github.com/Ralim/ts100/issues/87#issuecomment-349776478) 
That they talked to the author, which is what I was mostly going with?

perillamint, 2017-12-12:
Ah, I missed that point. In that case it would be OK.

matmemo, 2017-12-18:
Hi Ralim,

I downloaded the font and hacked up a python script to generate the 6x8 font data. It works quite good but there is a problem with it that I can't figure out at the moment. When compiled with the changed Font.c file the big 12px font gets corrupted. It still shows some chars but most of them become completely filled blocks on the display.

I'd like to send a pull request, but as it corrupts the normal font I don't know if that would be helpful. Maybe you can take a look at it in my cloned repo: https://github.com/matmemo/ts100/commit/794bfe3fdb22e9f18baa6d871cd2bce1b81fb5b5

The python script is found under ts100/Development Resources/fonts/gen_font6x8.py. It needs the numpy library and freetype-py bindings from: https://pypi.python.org/pypi/freetype-py

I copied the resulting data to Fonts.c with only some slight touchups in the comments following the char data.

Please let me know if you have any ideas about whats happening with the normal font, or if I should send a pull request.

Ralim, 2017-12-30:
Thanks @matmemo for this :)

closed 2017-12-30 by Ralim