Ralim/IronOS #149: Convert UniFont Over

Ralim, 2017-12-12:
To enable support for Chinese/Japanese/Korean, we need a new font.
Over in Issue #95, [Unifont](https://en.wikipedia.org/wiki/GNU_Unifont) in 16x16 resolution was identified as the ideal looking font.

Someone needs to convert this font map over to the format that is used by the OLED screen in the TS100.

Not sure when I'll have time, so leaving this open to anyone who wants the challenge.

perillamint, 2017-12-12:
GNU Unifont's Korean region looks somewhat ugly to me. I think Doongunmo font(https://github.com/perillamint/dkbfnts/blob/master/kimjoongtaefonts/H04.FNT) is much better than Unifont's one.

Ralim, 2017-12-12:
Honestly, I'm open to any fonts at this point, but I just don't have time to convert them over really.
I'll take all suggestions.

closed 2019-08-27 by Ralim