Ralim/IronOS #150: Tip not heating

19clarnoah, 2017-12-12:
This is an issue with the iron itself i believe, but i am using the most recent version of your firmware and i love it. It isn't my iron, but my tech teacher's, he lets me use it a lot. Problem is, the temperature of the tip stays between 21 and 28 degrees celsius. It refuses to heat up. I've done a few factory resets as well as reinstalling the firmware, still no heat. It was working today then suddenly the temperature just started dropping and it hasn't worked since. I was running at 310 degrees. I tried from voltages between 12 and 24, no difference. Bad tip?

19clarnoah, 2017-12-12:
I just got the firmware today, before it was the absolute stock version. Same heating issue occurred but if i waited about 5 minutes the heat came back. Now it's gone with no sign of return.

Ralim, 2017-12-13:
Hi,
Can you confirm that if you set the unit to the advanced idle screens if the temperature shows up as a logical ambient temperature?

Then when you start the heat up cycle, does the temperature stay correct but just not heat up?

If both of the above are true, then the feedback path from the tip is working so the header element is *probably* ok. But then the next most likely part to have failed is the power MOSFET.

One option might be to pop the back off the handle and have a quick look if anything has gone dodgy.
Also, check the smaller PCB that holds the STM32 is secured, I have had similar issues on my tester iron when that board was coming loose.

19clarnoah, 2017-12-13:
Yeah, I've got the advanced screens on and the temperature on the idle screen is between 21-28 degrees celcius. This is logical as the AC in the classroom has failed and it has been uncomfortably toasty in the room all day. Whenever i enter soldering mode, the same temperature is displayed. What you mentioned about the loose board seems highly logical to me, i'll check it out in class tomorrow and let you know what happens. Thanks for your reply! Btw, awesome job with the firmware! I never thought a soldering iron could be that advanced but hey here it is

Ralim, 2017-12-13:
Yeah I would say to check the little daughter board.
It sounds like for whatever reason the mosfet that switches on the tip heating isnt working. A quick visual might lead to more clues.

If you look at : https://ralimtek.com/images/TS100/schematicPWM.png Its basically something in that path somewhere that's not happy.

19clarnoah, 2017-12-19:
hiya, it's been a while. but I tested the mosfets, both of them, using continuity, diode voltage drop, and resistance. all 3 checked out fine on both mosfets. so I decided to test the continuity between tip connections A and B to positive and negative at the DC connector. B is continuous with negative and it makes sense because on the schematic for the entire circuit B is grounded. However, A is also continuous with negative. Looking at the schematic, I can't see how there could possibly be a direct, unobstructed path between A and negative. A and B are continuous with each other. Something dead in the circuit that is allowing A and B to have continuity with each other?

closed 2017-12-30 by Ralim

marvedjensen, 2018-07-09:
i have 2 irons with the same problem.   here is a little video of them 
https://www.youtube.com/watch?v=-Zbb4WM-FCg&t=199s