Ralim/IronOS #152: Scrolling should be faster

jonnieZG, 2017-12-13:
The new smooth scrolling looks great, but it is quite slow now. I think it should be at least 2x the current speed. So either the pause between the frames should be reduced, or 2 instead of 1 column-per-frame shift should be used. 

Sorry, I didn't study (yet) how the text display routine is implemented.

Ralim, 2017-12-13:
Yeah, easy fix (either option).
Easy enough to add too, just wasn't sure how others perceived the speed.

jonnieZG, 2017-12-13:
Yes, you should test both options, because I think reducing the pause might induce extra flickering.

federck, 2017-12-16:
Is also possible to unify the scrolling method?
Descriptions appears "by line" and disappers "by character".

jonnieZG, 2018-01-02:
Ok, the faster smooth scrolling is better now, but the letters now get torn in the middle. If you make a single shift by 2 instead of 1 px, and half the step time, I bet the tearing should significantly decrease.

Ralim, 2018-01-02:
Already testing that right now on my unit.
Will be pushing in next release :)

closed 2018-03-06 by jonnieZG