Ralim/IronOS #153: Wrong character index offset for C3 section

jonnieZG, 2017-12-13:
Characters in the 0xC3 sections start from 0x80, not from 0xA0 as in the 0xC2 section, so the code in Font.h should be 

```C++
		case 0xC3:
			index = (128) + (c);
			break;
```

instead of

```C++
		case 0xC3:
			index = (128-32) + (c); // THIS IS WRONG
			break;
```

Ralim, 2017-12-13:
huh, I added that as it fixed a language for me when I was testing. Can you tell me what language it broke so I can test further?

jonnieZG, 2017-12-13:
Sincerely, I don't know which language is affected by it, but it is obvious from Font.h that the C3 section starts from C380 (currently line 156), while C2 starts from C2A0 (line 121).

BTW, for a full Unicode/UTF map I use this http://www.utf8-chartable.de/unicode-utf8-table.pl.

closed 2018-01-02 by jonnieZG