Ralim/IronOS #154: Bad characters

jonnieZG, 2017-12-13:
Some characters, such as "ć" (as much as I can see for the Croatian translation) is displayed as a filled rectangle.

@Mrkvozrout, you said you too have the same issue with some characters in the high pages. From what I saw in issue #132, your EEPROM might be worn out and faulty, but my unit is brand new, and I strongly disbelieve that this is causing problems in my case. While all the characters look fine, "ć" is displayed as a filled rectangle. 

UTF-8 code for "ć" is c4 87, while "č" with the code c4 8d is displayed fine. I suspected that the reason might have been some signed/unsigned overflow, but as much as I can see from the STM32 documentation, char type is unsigned there... That's really strange. 

Does anyone else have an idea what is causing this issue?

Ralim, 2017-12-13:
I doubt its the EEPROM, usually this is just a character pointer going wrong somewhere.

char type is uint8_t under the hood, and UTF is encoded in multibyte sequences.

Which language build is this appearing in and ill try and narrow it down :)

jonnieZG, 2017-12-13:
Currently the problems were reported in Polish (for string "U0100 0Ā 1ā 2Ă 3ă 4Ą 5ą 6Ć 7ć 8Ĉ 9ĉ AĊ Bċ CČ Dč EĎ Fď") and for Croatian language. The Croatian translation does not use all the diacritic characters, so you can use this string to test all of them: "ČčĆćĐđŠšŽž".

jonnieZG, 2017-12-21:
Possible causes also discussed in issue #170 .

jonnieZG, 2017-12-31:
Works fine now! May I know what was causing the problem?

Ralim, 2018-01-01:
An error where I had miscalculated the memory location for the settings structure and it was overwriting some of the font table.

closed 2018-01-02 by jonnieZG

Mrkvozrout, 2018-01-03:
Updated czech translation with diacritics in #185. Works fine after the fix.