Ralim/IronOS #155: Reading the accelerometer value and display on LCD

riccardopittini, 2017-12-14:
Hi
I want to implement the temperature setting with an algorithm that uses the accelerometer.
I started by 
#define ACCELDEBUG 1

Then in the main I added the following lines (and changed the function from private to public)
taskENTER_CRITICAL();
uint8_t tempVar1 = accel.I2C_RegisterRead(OUT_X_MSB_REG);
uint8_t tempVar2 = accel.I2C_RegisterRead(OUT_X_LSB_REG);
taskEXIT_CRITICAL();

The problem is when I print on the LCD the values for MSB and LSB (I'm reading the X-axis), they seem garbage. MSB is always 0 or 255 and LSB varies almost randomly independently on the position of the soldering iron.
I tried, instead of reading the registers, to print the values stored in the "original code" (avgx, avgy and avgz) but I could not see any reasonable values.

Is there a quick and easy way of reading X,Y,Z axis values, display them properly on the LCD and then use them as input for an algorithm?

Thanks a lot for the Help, if the thing I have in mind works on this iron I will propose a new method for setting the temperature (I previously implemented the algorithm on a soldering iron based on a PIC16).

Cheers
Riccardo

riccardopittini, 2017-12-25:
No one has any suggestion? :)

Ralim, 2017-12-26:
Hey,
Sorry I missed this.
If you change it to `ACCELDEBUG 1` that should be all that is required to print out the acceleration values to the LCD screen. You want to look in the code at `startMOVTask` as it processes all the acceleration values etc.

The logic here is that normally the UI and the acceleration data is handled seperately, and since they share an I2C bus we have to be careful to ensure they dont get into a fight or lockups over who has the bus. Thus the task CRITICAL sections around I2C access.

riccardopittini, 2018-01-03:
Hi Ben

Thanks for the reply. I already set ACCELDEBUG 1 (it was the first thing I did). My problem is that the values I read they look allover the place (they do not make sense at all or do not correspond to an angular position).

I added also lines like: 
		lcd.print("x:");
		lcd.printNumber(avgx, 5);
In the MovTask so I can read the XYZ measurements on the display.
e.g. For the X values, I see the numbers varying from 64000 to about 3000, without giving real sense (not coherent with the xyz position of the stylus). Seems when I move it that it varies, and then the values goes back to zero or close to it (65500).

Have you tried yourself to set ACCELDEBUG 1 and verify that the values that you read on the display make sense?

I simply want to get the angular position of the 3axis to make an algorithm to set the temperature (an simple acos function that you can set the temperature with a movement of the iron - if you are interested I can explain it more, I previously implemented it on a PIC16F with an analog accelerator).

Would be great if you have a comment on this issue I'm experiencing!

Ralim, 2018-04-27:
Sorry I completely missed this.

Rememberng that the values are the forces being experience, when the iron is aligned to an axis (X,Y,Z) you will only see a reading in that one axis and others will be close to zero as you have observed. 
There is no scaling being done in the software for this numbers at all so what you are seeing sounds right.

3 of the numbers are live readings, and the 4th is a max(sum(x,y,z)) value.