Ralim/IronOS #158: Formatting

Mrkvozrout, 2017-12-15:
Hi, I wanted to point out, that the current formatting of the source codes is not good and it is getting worse.

Look at the [currently last commit](https://github.com/Ralim/ts100/commit/4c0ba725421190d4f3ef4d477e751db7e4c59468) e.g. files `gui.cpp` or `Translation.c` - what a mess!

With this it is hard to read it, maintain it, merge it and so on.

There could be defined some readable formatter with common best practise rules like:
- indent with spaces, not tabs
- do not trim lines to 80 characters as it does not make sense with today's wide monitors
- blank line between functions, after directive groups, ...

Ralim, 2017-12-16:
Typically I was letting the eclipse formatter handle it, which sometimes does a terrible job.
I'm in the process of setting up a pre-commit clang-format rule for the code base :)

Mrkvozrout, 2017-12-16:
Interesting, also well configured Eclipse formatter would be ok. Thanks for solution :)

Mrkvozrout, 2017-12-16:
I think the eclipse formatter would be better from the point of view that what you see is what you commit. Also formatting in general should not be done on whole files, just your changes. What do you think @Ralim ?

Ralim, 2017-12-20:
Not sure what would be the best ideally, normally I was trying to just run the eclipse formatter on the whole file before commit. However, that was getting messed up with the different `#defines` before. 
Should be better now that they are done at compile time rather than in a file.

Mrkvozrout, 2017-12-20:
The worst thing is the 80 chars wide lines - I have never seen splitting 2 word comment into two 1 word comments just because of 80 char limitation... until now :D. I think the programmer should do most of the formatting while writing the code to make it readable, Even if it does not 100% follow any formatter rules. And formatting of whole file - if ever- should be committed separately because then if you want to look back in commit history, what someone changed and how, the changes are lost because the whole file is highlighted. These are my insights :)

Ralim, 2017-12-20:
Hahah, yeah 80 char lines can be annoying. Though it does make it nicer when reading on a mobile device :D 

That's correct, but if everyone formats the same then the problem also disappears. My main issue with people manually formatting is that they tend to use spaces :P

Mrkvozrout, 2017-12-21:
Do you mean spaces vs tabs? If you are not writing a makefile, spaces are the way to go :D Tabs mostly break indentation and formatting in general, because they are **not handled uniquely** by IDEs, text editors, web pages... one shows them as 2 spaces the othe as 4 spaces and the othe as 8 spaces. Moreover if someone tabs get mixed with spaces one programmer sees it ok but other other has it broken because his editor shows tabs as 4 spaces instead of 2... tabs are evil, 2 spaces are common ractise, trust me, I come from SW industry :)

Ralim, 2017-12-21:
I've only been in the software industry since ~2009. 
I am well aware of the tabs vs spaces issue among people :D I was also basically laying bait to see where you stand on the issue.

Since this software is mostly developed in eclipse and other editors that tend to stick to the 4 spaces per tab rule it makes it easy. Also if every file is only indented using tabs than the actual value is of less importance.

It's a big issue that can be argued many, many ways. I personally prefer tabs in indentation independent languages because then my indentation *amount* is up to me, and tabs convey the intent of indentation, rather than the actual width.

Mrkvozrout, 2017-12-21:
Yeah I am aware that this is one of "those issues" :D

Well I still use the TAB and SHIFT+TAB to indent code, but I have the IDE configured to use spaces so it still is 1 TAB = 1 indent, but uses spaces so it is rendered the same way everywhere. And you really do not look at what whitespace is used, just want to have the indentation right.

But in the end, this is your repository so I should (and will) honor your conventions ;)

Ralim, 2017-12-21:
Haha yeah,
So long as the indentation looks good I generally don't overly care, though I always ensure tab = 4 spaces since it seems to be the most common across the IDE's I use.

Honestly, so long as people roughly match whats nearby I dont stress *too* much.
Though I'm going to see about setting up a nice formatter :)

closed 2019-08-27 by Ralim