Ralim/IronOS #159: Italian bad characters and some other issues

federck, 2017-12-15:
All accented vocals (such as à and ù) are dispayed as uppercase letters.
Also «ADVSLD» option doesn't provide any description and «CAV VIN» shows "Display detailed information while soldering" description.

Is possible to decrease time for description to come out?

Really big thanks for your great work,
Federico

Ralim, 2017-12-16:
Thanks for flagging this with me, seems to be a few font issues for me to sort through on this latest release :/

Yes I will be reducing the time for the descriptions too :)

Mrkvozrout, 2017-12-16:
This could be the wrong offset mentioned in #153 (because difference between lower and upper case is 32)

Mrkvozrout, 2017-12-16:
The description positioning wrong in several other languages. See #166

closed 2017-12-24 by Ralim