Ralim/IronOS #167: Logo is lost on 2.0 to 2.01 upgrade

manevich, 2017-12-17:
I had to re-flash logo on my iron, while according to https://github.com/Ralim/ts100/wiki/Logo-Editor logo should stay intact on upgrade. Setting are lost too, btw.

I think B800-BBFF range should be excluded from firmware HEX file to keep logo intact.

Ralim, 2017-12-24:
This was caused by a firmware bug that I made ages ago, and we just finally cross that address range in code size. The next version of the firmware will completely fix this for you :) However you will need to re-generate your bootup logo file sadly :/

manevich, 2017-12-24:
Thanks for looking into this.
If I understand correctly, I have damaged firmware by flashing logo hex file?
Nothing weird happened so far and iron is completely functional, but I still going to re-flash firmware to be sure.

Ralim, 2017-12-24:
Yeah, it seems to overwrite the font table at the moment. Nothing overly detrimental (and there are some hardware protections in the unit so don't panic). But probably the best idea until I can push out at the least a new alpha.

Ralim, 2017-12-30:
This is fixed in 2.02 :)

closed 2018-01-09 by Ralim