Ralim/IronOS #168: Use Y/N instead of T/F

ulab, 2017-12-17:
It took me a moment to realize what the T/F settings meant... Perhaps it's better to use Y/N (Yes/No) instead?

Mrkvozrout, 2017-12-18:
@ulab Hi, do you mean english or other language? This is currently up to each language translation (translator's choise).

ulab, 2017-12-18:
English. I have installed the firmware without any language extension. (I'm still guessing T stands for True and F for False? :-))

This is something coders might realize, but for the "common people", I think Yes and No might be easier.

It's just a suggestion since it was the first thing that came to my mind yesterday.

Mrkvozrout, 2017-12-18:
Thanks to bring in your opinion and (re)start the discussion.

It is hard to decide what indication to choose (true/false, yes/no, +/-...). I think the commonly most realizable would be to implement ✓✗ characters. I will try to play with it and share results (if someone wouldn't be faster :) ).

matmemo, 2017-12-22:
I had a similar feeling as ulab when I did the german translation. In english it's kind of ok for me because it feels quite natural in a technical context if you've done some programming. But translated to german it really get's awkward. The English letters feel out of place and the german equivalent W/F isn't used in devices I know of and I doubt anybody would understand these letters at first glance anyway.

What I'm getting at is my favourite solution up to now. With the new double line mode for the settings it's posible to simply pose the concerned setting as a question, ending with a question mark. After that it's quite clear that I have to answer Yes or No (J/N in german).

Well that's just my two cents on the matter and I'd really like to know what others are thinking about that idea.

Mrkvozrout, 2017-12-23:
@matmemo I think that is really good solution. In my czech translation I bet on when users don't not know what the values are, they wait (or stare in confusion :D) until the description shows and they can read what thhe options mean.

What do you think about the ✓✗ chars?

Rockman18, 2017-12-23:
I've pushed a pull request #177 proposing a checkbox instead of T/F or Y/N

SilverSho0t, 2017-12-27:
I think ✓✗ characters are a very good solution. It will simplify the menu.

MichMich, 2017-12-30:
My first though would be ✓✗as well.

wortex1979, 2017-12-30:
i think ✓✗  characters is very good

Rockman18, 2017-12-30:
Pull Request #177 add a checkbox intead of True/False. This PR has been merged in master last week but not compiled/released yet. The checkboxes look like that : [image] [image]

I've compiled a V2.02-alpha.1 in English to share it with you. Tell me your opinion for thoses checkboxes.

[TS100_v2.02-alpha.1_EN.zip](https://github.com/Ralim/ts100/files/1594421/TS100_v2.02-alpha.1_EN.zip)

SilverSho0t, 2017-12-30:
Very nice @Rockman18 much more understandable & user friendly ;)

MichMich, 2017-12-30:
Look much better than the T/F. Awesome. You might want to add a bit of margin at the bottom so that it is better aligned to the inter of the screen.

Mrkvozrout, 2018-01-02:
@Rockman18 good work, you were faster :)

I offer my pull request #183 with updated checkboxes with discussed ✓✗ chars. I think it looks neater and even more understandable (as the X looks more descriptive then current empty chackbox).

What do you guys think? (see PR for photos)

PR #183 (2018-03-03): Use unicode chars U2713/U2717 as true/false in menu
Based on latest master (with already replaced T/F chars for check boxes) I tweaked the T/F chars to be:
- true - ✓ - U2713 - check sign
- false -✗ - U2717 - ballot sign

I think it looks neater and even more understandable (as the X looks more descriptive then current empty checkbox):
[image]
[image]

resolves #168
closed 2018-03-03 by Ralim