Ralim/IronOS #180: Calibration ambiant temperature

SilverSho0t, 2017-12-27:
Hello I have a little problem with your software [else I really love it ;)], when I install the software the temperature displayed is 28°C while my room temp. is 21-22°C, so I do a calibration but after it show me 18°C, so I have a question.
Can you add, when we select calibration, a menu where we select the current room temperature to do a good calibration ?

Thanks in advance & thanks a lot for what your doing here.
Sean.

Ralim, 2017-12-27:
The unit uses the temperature sensor in the handle for the calibration offset. 
Can I ask if the unit had been in your hand or in use when you performed the offset?

Basically, there is a small temperature sensor (I think at TMP36 series) in the handle to allow for cold junction temperature offset during use. In calibration the unit samples an average of readings from this unit and compares this with the tip reading, storing the offset. This is then used to adjust the offset that occurs from tip to tip.

Ralim, 2017-12-29:
The sensor in the handle is +-0.1C from the factory. Almost certainly its measuring correctly.
The linear approximation I'm using for the tip measurement is based on operating temperatures, so I would not be overly surprised if its out by a few C at the cold end (ie not heated). Can you check the temperature accuracy of your unit when the tip is heated?

On my unit its cold temperature sense is out by about 2C but when hot its around 2C of error as well, which is a small amount (2C/400C =0.5% error).

Remember you are calibrating for hot temp, not cold temp when you are calibrating a soldering iron. The only reason the calibration is done at ambient is that we already have the temperature sensor in the handle conveniently, and not all users have a thermocouple around to calibrate against 😄

SilverSho0t, 2017-12-30:
I have no more calibration problems with the update posted by @Rockman18 here #168.
Thanks anyway for your help.

closed 2017-12-30 by SilverSho0t