Ralim/IronOS #182: Include some example boot up logo files

Lenbok, 2018-01-02:
It would be nice if there were a few nice example boot up logo files that the lazy/unartistic among us could just directly flash :-).

Ralim, 2018-01-02:
If anyone out there has any to donate I'm happy to add them to the repository.
Haven't really made any myself that I like :)

JohnEdwa, 2018-01-05:
Nothing beats some good old [Chinglish from the marketing material](//img.banggood.com/images/upload/2015/12/552/1.jpg).

[image]

[image]

Sadly "After 10000 times Striking Test Still Steadfast" and "When You're Not Here, He's Waiting For you Quietly" are too long to fit properly ;(

Schattenbanane, 2018-01-07:
I spend some time creating something clean and vaguely "officially-looking".
[image]
If somebody wants to use it they are welcome to do so.

wortex1979, 2018-01-08:
schattenbanane very good logo

jonnieZG, 2018-01-08:
Here's mine. However, I'm not sure whether is it legal to use their logo in the custom software. 

[image]

SmayK, 2018-03-16:
here is my image idea
[image]

federck, 2018-03-31:
Mine here:

[image]

[image]

closed 2019-08-27 by Ralim

River-b, 2022-02-12:
would be awesome if all these other Custom logos could go into central place in Github so people could find them easier to browse for one they like.

River-b, 2022-02-12:
> Mine here:
> 
> [image]
> 
> [image]
> 
> [image]

I like your Boot Logos I'm going to convert them to DFU  for Pinecil :)    do you have any more let me know

River-b, 2022-02-13:
This is my first Logo Art ever 
Doggie Bone
[image]

I could not attach the DFU in here, but just convert that image using the Python script in the Boot Logos folder.