Ralim/IronOS #187: Always show temperature and keep display on in idle mode if the tip is still hot.

JohnEdwa, 2018-01-05:
Right now if I shut the iron off and cancel out of the cooldown screen, it'll go back to the idle screen that doesn't show the tip temperature. Even if you have 'Detailed Idle Screen' toggled, it'll shortly shut the screen off even when the tip is still dangerously hot (190C in my case).

Id's suggest always disabling screen sleep if the tip is over 50C and implementing the tip temperature warning to the default idle screen as well.   
Maybe replace the "soldering mode" icon with the exclamation triangle or simply with the tip display?

Ralim, 2018-03-03:
This should be fixed in the form that it wont let the screen goto sleep etc.
I haven't added the tip display yet. Not sure the best easy way to do this without encoding extra graphics yet.

JohnEdwa, 2018-03-11:
I'd just write the temperature where the 'start soldering' icon usually is until it got cool enough to be save (so... under 50C I guess). 
Something like this:
[image]

Btw, do you have the icons, fonts and other resources available somewhere so I can start making better-looking mockups :P ?

Ralim, 2018-03-11:
Yeah thats basically what im looking at doing now.

Not really, some of the icons are available in the issues that created them, for the ones I created i just draw them using a binary-> hex converter and time :)

JohnEdwa, 2018-03-11:
Oof, no kidding. When you try to convert them back to [XBitmaps](https://en.wikipedia.org/wiki/X_BitMap), this is your 'WarningBlock24': [image] Instead of left-to-right, you encode them vertically, split into two, that are mirrored and in the wrong order :P

Oh well, I got to hone my almost none-existant python skills again and managed to construct a script that converts them back successfully, so here they are:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

If you could stick these into the 'Development Resources' folder, and save any new icons you come up with, I'd appreciate it.

Ralim, 2018-03-11:
Yeah they are in the native format for the screen, which is odd if your not used to B&W screens.

Thank you :)

Do you want to add them using a pull request? If not I'll add them :)

Thanks for converting them back :)

Also there should now be a public branch that implements this (drawing the temp), I can compile a test firmware if you would like.

JohnEdwa, 2018-03-11:
You can add them, it's faster that way :)

And a test build would be neat, but you really should release a new alpha ASAP to fix #223 especially as the old stable isn't compatible with the new accelerometer.

Ralim, 2018-03-12:
@JohnEdwa I'm gong to try and fix up the #214 concerns and then push one out. 😄 
Doesnt help that it seems most commenters are on the other half of the world, so I wake up to notifications :)

Ralim, 2018-03-14:
@JohnEdwa Pushed a new release tester out, if no-one has any big issues ill make it a full release and then go from there.

PR #226 (2018-04-27): Add showing the tip temp on idle
This shows the tip temp on the simple home screen if the temp is > 50C.

This removes the tip temp warning for all users, as now everyone can see the tip temp.

Fixes #187