Ralim/IronOS #188: Splashscreen logo, wrong initial orientation & mirroring.

JohnEdwa, 2018-01-05:
For the first second, the logo is always displayed in right handed orientation, no matter what setting or orientation the iron is. If it then notices it should be displayed the other way around, it mirrors it vertically (like it does for the idle screen) instead of rotating it properly.

Ralim, 2018-01-09:
The first second delay is because of the startup time of the accelerometer, its output is unstable for when the unit is first powered up. 
It should show up in left-handed orientation if you have it set to left-handed only (not auto) in the settings. That's a bug I'll fix.

Ah right, I'll fix the rotation issue soon too.

Ralim, 2018-03-14:
This *should* be fixed in 2.03-rc2

closed 2018-03-14 by Ralim

JohnEdwa, 2018-03-14:
Confirmed - mirroring is fixed, as is the initial orientation when set to L.