Ralim/IronOS #190: Include option to have custom idle screen

Journesaladbar, 2018-01-06:
First off thank you so much for sharing your version of this firmware. 
The first thing I did was work on custom boot-up logos inspired by retro games to use on my Ts100.
[image]
Now I played with a possible idle logo and would love to use the editor and create a custom idle screen the same way (shown below).
[image]
Lastly, I felt like the detailed idle screen would work better as the detailed soldering screen. 
Again, thank you for this version of the firmware because it is pretty feature rich!

Ralim, 2018-01-09:
Good ideas!

Custom idle screens should definitely be an option to add in the next release :)
Should be able to pack them in with the bootup logo to save space. I will need to think on how to do this nicely in the creation tools though.

Gifs are a bit harder, just because it adds a lot more complexity to the system and requires storing a larger number of frames for the animation. So that might be a future expansion.

That would be a good idea to add to future builds, the warning would probably be appreciated if you forget you have dismissed it.

Ralim, 2018-05-10:
This could be packed into the changes that can come with #292.

discip, 2022-04-23:
@Journesaladbar
Does the latest release satisfy your request?
If so, please consider closing this issue. 😊

thanks

River-Mochi, 2023-03-23:
I don't see option to have custom Idle screens.  only to have custom boot logos on bootup.
when my Pinecil V1 goes to sleep  it says "sleeping"

not sure how I  would go about changing the idle  screen to custom based on the tools we have.

Ralim, 2023-04-01:
The only way to change it would be to change the code.
We could have an option to show the bootup logo instead when its asleep; but its preferrable to keep the OLED off to avoid burn in imho.

River-Mochi, 2023-04-01:
> but its preferrable to keep the OLED off to avoid burn in imho.

Ageed.  I like that OLED goes to black screen and then wakes up when i pick it up.  It's about $5 with shipping everytime you need to change your OLED screen. depending on brightness picked, how often you use the screen,  I've seen it as low as  1 year for some people but others have it 3 years no issues.

this doesn't look like a good idea,  since you mentioned wanting a  balanced option to  let the screen rest during idle so that the OLED doesn't burn out faster, perhaps this ticket should be **closed**.  Note it's from 2018  and **5 year  old ticket.**

discip, 2023-04-01:
> but its preferrable to keep the OLED off to avoid burn in imho.

Due to the weight of this statement, I am closing this issue.

closed 2023-04-01 by discip